PX4/PX4-Autopilot #3167: Parameter parsing on NuttX update

LorenzMeier, 2015-11-09:
This one is interesting:

The mTECS controller is not enabled by default, but linked. Its using the C++ param system. Its parameters should be linked as usual.

```
error finding param: MT_ENABLED
error finding param: MT_THR_MIN
error finding param: MT_THR_MAX
error finding param: MT_THR_I_MAX
error finding param: MT_THR_FF
error finding param: MT_THR_P
error finding param: MT_THR_I
error finding param: MT_THR_OFF
error finding param: MT_PIT_MIN
error finding param: MT_PIT_MAX
error finding param: MT_PIT_I_MAX
error finding param: MT_PIT_FF
error finding param: MT_PIT_P
error finding param: MT_PIT_I
error finding param: MT_PIT_OFF
error finding param: MT_FPA_P
error finding param: MT_FPA_D
error finding param: MT_FPA_D_LP
error finding param: MT_FPA_MIN
error finding param: MT_FPA_MAX
.error finding param: MT_ACC_P
error finding param: MT_ACC_D
error finding param: MT_ACC_D_LP
error finding param: MT_ACC_MIN
error finding param: MT_ACC_MAX
error finding param: MT_FPA_LP
error finding param: MT_ALT_LP
error finding param: MT_A_LP
error finding param: MT_AD_LP
```

davids5, 2015-11-09:
@LorenzMeier 

1) When you say "Not enabled" are you saying it is not listed in cmake/configs/nuttx_px4fmu-v2_default.cmake?

 The change to the px_generate_params.py to limit the scope to only the included modules would cause this. - We may need a  section like forced_included_parameters and list the modules, in there that should be included in the scope even if not listed as included.

2) Where are the errors listed?

LorenzMeier, 2015-11-10:
You need to set the autostart to 3033 and watch the console. And we need to do param scanning recursively in all directories below the module dir, not just the module dir. Should be a trivial change.

davids5, 2015-11-10:
Fixed in https://github.com/PX4/Firmware/commit/cb44789a228ce1da1d28b49a50cee7c83e078b1a

closed 2015-11-10 by davids5