PX4/PX4-Autopilot #3168: Unify Estimator/ Controller Framework

jgoppert, 2015-11-10:
Most of us are using this as a research platform and it really is rather difficult to setup a new estimator/ controller to compare against existing methods. I'd like to propose that we create a unifying framework that just requires the user to derive from a base class and implement a few functions.
This would allow even newcomers to quickly test out their ideas. For example, this would be the end user code:

https://github.com/dronecrew/Firmware/blob/lqg_quad/src/modules/lqg_quad/BlockEst.hpp
https://github.com/dronecrew/Firmware/blob/lqg_quad/src/modules/lqg_quad/BlockEst.cpp

Proposed Base Class for attitude/position estimators:
https://github.com/dronecrew/Firmware/blob/lqg_quad/src/modules/controllib/estimators/BlockAttPosEst.hpp
https://github.com/dronecrew/Firmware/blob/lqg_quad/src/modules/controllib/estimators/BlockAttPosEst.cpp

LorenzMeier, 2015-11-10:
@jgoppert I guess its as always a matter of preference. I also regularly meet developers who feel choked by having to use a base class and just want to pub/sub. I think within the scope of our core dev efforts it however makes certainly sense to put more effort into consolidating onto frameworks.

jgoppert, 2017-04-28:
I think we have shown just having the standardized messages is good enough.

closed 2017-04-28 by jgoppert