PX4/PX4-Autopilot #3179: Significant rotation imprecision in get_rot_matrix()

LorenzMeier, 2015-11-13:
This function:
https://github.com/PX4/Firmware/blob/16b6b3caa2e152888bc942dbe723cfab3320d526/src/lib/conversion/rotation.cpp#L43-L51

Returns this for identity:

```
1.000, 0.000, -0.025,
0.000, 1.000, 0.011,
0.025, -0.011, 1.000
```

tumbili, 2015-11-23:
@LorenzMeier I cannot reproduce this in SITL. Did you try this on NuttX?

LorenzMeier, 2015-11-23:
yep, this was NuttX. So its probably the ARM Matrix stuff.

tumbili, 2015-11-23:
'''
float cp = cosf(pitch);
		float sp = sinf(pitch);
		float sr = sinf(roll);
		float cr = cosf(roll);
		float sy = sinf(yaw);
		float cy = cosf(yaw);

		data[0][0] = cp * cy;
		data[0][1] = (sr * sp * cy) - (cr * sy);
		data[0][2] = (cr * sp * cy) + (sr * sy);
		data[1][0] = cp * sy;
		data[1][1] = (sr * sp * sy) + (cr * cy);
		data[1][2] = (cr * sp * sy) - (sr * cy);
		data[2][0] = -sp;
		data[2][1] = sr * cp;
		data[2][2] = cr * cp;
'''

This is how the function looks. Does that mean that the trigonometric functions are not performing well? Or am I missing something?

tumbili, 2015-11-23:
I can't see any arm specific functions involved.

LorenzMeier, 2015-11-26:
@pavel-kirienko I think you had a rotation offset set from the level calibration routine. All unit tests pass.

closed 2015-11-26 by LorenzMeier