PX4/PX4-Autopilot #3188: MC: Enable altitude hold using Lidar as positioning source

LorenzMeier, 2015-11-15:


LorenzMeier, 2015-12-02:
@AndreasAntener I think you're doing this anyway, right?

AndreasAntener, 2015-12-02:
Not exactly. Depends a bit what you mean here, but generally we find using the lidar for altitude hold outdoors would have a lot of problems with changing underground.

ecmnet, 2016-01-06:
My comment on this: Using SF10A Lightware LIDAR on my own FlightStack gave quite good altitude hold results outdoors (+/- 5cm); i experienced no underground issues. Since I decided to switch to the PX4 flight-stack, I wonder now, how to integrate SF10A to PX4 using I2C connection... guess I have to write a driver for it...right?

ecmnet, 2016-01-06:
.. +/-5cm was on grass, on more leveled ground even better.

AndreasAntener, 2016-01-25:
@ecmnet only just now noticed your comment on this. My concern about the "changing underground" was not related to the lidar performance on different surfaces, but about that we don't know the terrain therefore you can only hold altitude above ground but not absolute altitude hold (at least not without a terrain model).

Btw, did you compare the performance of your SF10A on grass vs. tar? We found that the PulsedLight Lidar looses at least 15m of range over tar and I was wondering how the Lightware behaves.

ecmnet, 2016-01-25:
@AndreasAntener ok. Haven't got this. Of course you're right. Concerning the SF10A performance on tar, I'll take that with me for the next flight outdoors

LorenzMeier, 2016-04-01:
Done on LPE

closed 2016-04-01 by LorenzMeier