PX4/PX4-Autopilot #3190: UAVCAN -Wfatal error

swatza, 2015-11-16:
[image]

Does anyone have a solution to the above problem?

mhkabir, 2015-11-16:
What GCC version is this on?

swatza, 2015-11-16:
Not on a linux machine: mingw32

pavel-kirienko, 2015-11-16:
@swatza GCC version can be obtained by executing `gcc --version` (add necessary toolchain prefix before `gcc`, e.g. `arm-none-eabi-gcc`).

This issue should be fixed by https://github.com/PX4/Firmware/pull/3192, but I'm still curious about what GCC version triggers that error.

swatza, 2015-11-16:
4.6.2

pavel-kirienko, 2015-11-16:
I'm not sure if GCC 4.7 is still supported. The toolchain installation pages recommend 4.8.x. Anyway, you can use the patch I have linked above, it should fix the build even if you're keen to stick to 4.7. 

Feel free to re-open if it didn't help.

PR #3192 (2015-11-18): Removed an excessive template disambiguator
...in the virtual CAN driver class. This fixes #3190.
closed 2015-11-16 by pavel-kirienko