PX4/PX4-Autopilot #3198: Tablet-only flight

LorenzMeier, 2015-11-17:
We should work towards demonstrating tablet-only flight for multicopters, VTOL and planes in AUTO. We should address all usability issue with that along the way.

Important configuration options to set:

  * COM_DISARM_LAND: Auto-disarm in seconds after landing. Set to 3 for testing

tnedev, 2015-11-18:
We are now flying tablet only with a fixed wing and without lidar. No general stoppers except fragility of the takeoff/land with a few glitches. 
Hopefully I will be able to check the new improvements in a few hours.

tnedev, 2015-11-19:
@LorenzMeier  I think a better switching between the different command interfaces will be very useful. Mainly that you could have your RC ON while using the tablet and fast switch to RC control if needed.

LorenzMeier, 2015-12-02:
@tnedev Thanks, that's helpful feedback.

LorenzMeier, 2016-01-06:
We have this now covered, you can fly with only a tablet.

closed 2016-01-06 by LorenzMeier