PX4/PX4-Autopilot #3207: Support a command to take off and hold that position / land

LorenzMeier, 2015-11-18:
@tumbili We desperately need a simple takeoff command which puts the system into flight state. Mostly for soak-testing in SITL, but probably also for our ongoing efforts to allow tablet-flying. We also need the equivalent land command.

Part of that will involve adjusting the controller to do a better job during takeoff and landing, but the first step will be to add the logic. We can discuss how to best do this.

The priority is critical since we need this for our integration efforts with Qualcomm. They have right now no simple way to put the system into flying state from the command line.

LorenzMeier, 2015-11-18:
The goal is that

```
make posix jmavsim
```

Will end up with a hovering system. That way it will become immediately apparent if the system fails to hover.

tumbili, 2015-11-23:
@LorenzMeier Ping me if this is still critical, I've seen the "commander arm" command which is not what this should do but a first step.

LorenzMeier, 2015-11-23:
Implemented "commander takeoff" on my SITL branch. Closing.

closed 2015-11-23 by LorenzMeier