PX4/PX4-Autopilot #3211: Add back true map caching

LorenzMeier, 2015-11-18:
The current approach with zooming in and out only works for a small flying field and even there not reliably. We need a way to load all tiles in the current view down to the max. zoom level on  button click.

@dogmaphobic How hard / easy is that with our map backend?

dogmaphobic, 2015-11-18:
We had this conversation earlier :)

>dogmaphobic Sep 26 18:28
>You need to go to where you want to “cache” and scroll around at the magnification you want. If you are planing a trip to a region that’s 5,000 square miles (like me next week), you need to scroll over the entire region at all the magnifications you may want. That’s basically impractical. It would take days scrolling around.

>DonLakeFlyer Sep 26 18:33
>That seems like an edge case though. Basically you have no internet across that whole period of time and space, which would seem uncommon. I think starting with the simpler, no ui, fix the cache approach would be the better place to start. Then get fancy later.

It's not exactly simple. We're using QtLocation's default cache mechanism. To cause it to cache something, we need to actually *display* a tile on the screen. To automate a *region caching* we would need to programmatically pan around a defined region at a specified zoom level. It would probably be simpler to create our own cache mechanism so we can load the tiles in the background instead. I know how to do both. The first is easier but weird as the UI would be showing the map move around while *caching*. The later would be cleaner and faster but a whole lot of work.

LorenzMeier, 2015-11-18:
I think having the view move around is perfectly fine. It helps to show the user that its "doing something".

DonLakeFlyer, 2015-11-18:
This is in the firmware repo :)

I still think 5000 square miles is an edge case. You would cache smaller individiual sections. For moving the view around, you not only need to do that, but wait for all tiles to finish.

dogmaphobic, 2015-11-18:
> This is in the firmware repo :)

Right! :) I didn't notice until you pointed out.

> For moving the view around, you not only need to do that, but wait for all tiles to finish.

That's why it would be quite a bit slower.

DonLakeFlyer, 2015-11-18:
Still seems like a decent way to go as a start. Only speed difference will be draw time, since download time will be the same. I think long term our own caching backend may be needed. This would give us better information like being able to show what is currently cached and so forth, as well as whether you are filling the cache or not, cache size, etc.

LorenzMeier, 2015-11-18:
yeah, l think starting with something basic is perfectly fine. Most users won't mind and the fact that things are moving will actually make them feel good about the level of progress and activity.

LorenzMeier, 2015-11-19:
Epic fail - registered it properly now:
https://github.com/mavlink/qgroundcontrol/issues/2241

closed 2015-11-19 by LorenzMeier