PX4/PX4-Autopilot #3212: mpu6000 driver:Investigate if 'test' function causes register checking to find a false faults

davids5, 2015-11-18:
While developing  https://github.com/PX4/Firmware/commit/51a7a51d9f0ed290596030d31b16e37ba4fd323d

I noticed the the register checking fails on a test command. 

This needs to be debugged

dagar, 2017-04-11:
Still an issue?

davids5, 2017-04-11:
I have not tested it on latest master.

davids5, 2017-04-11:
@dagar - yet it was. See https://github.com/PX4/Firmware/pull/7024

dagar, 2017-04-11:
Was mpu9250 copied from mpu6000. Any overlap here? https://github.com/PX4/Firmware/issues/5446

davids5, 2017-04-11:
@dagar - yes - I would think it is the same issue. I did a pass on the 9250 in the PR, but I am not set up to test it. I asked @pkocmoud if he could give it a go. Do you have HW with the 9250 you can check it on?

davids5, 2017-04-11:
@dagar, @LorenzMeier 

One more thing. The root cause was not the test. It was the reset after the test. Now there is a drop out of collecting samples during the test for ~30 ms. Will that be an issue?

dagar, 2017-04-12:
Yes, I have a pixhawk 2.1 I could use for testing.

closed 2017-04-14 by LorenzMeier