PX4/PX4-Autopilot #3217: GPS spike detector

LorenzMeier, 2015-11-19:
We will need a GPS spike detector which does several things and probably can be a standalone thing like the terrain estimator. Maybe this is also something LPE already handles.

  * It detects excess acceleration suggested by GPS
  * It estimates the resulting velocity and position offset
  * It allows to remove that offset over some time as the GPS estimate slowly recovers / comes back

LorenzMeier, 2015-11-19:
Log with spike happening:
http://logs.uaventure.com/view/Cp3XkZ2EAvjA4W4xi4sRDk#Velocity_PLOT

kd0aij, 2015-11-20:
@LorenzMeier It looks to me as though the velocity plot is correct. Also, I'd appreciate an answer to the question I asked in https://github.com/PX4/Firmware/issues/2790#issuecomment-157434584; if I'm right that log exposes a bug in the altitude control logic.

LorenzMeier, 2015-11-20:
@tumbili Could you have a look? Is this what you fixed here? https://github.com/PX4/Firmware/pull/3229

LorenzMeier, 2016-02-18:
Will be done in EKF2.

closed 2016-02-18 by LorenzMeier