PX4/PX4-Autopilot #3220: UAVCAN ESC Enumeration does not seem to reset node ids

LorenzMeier, 2015-11-19:
Currently it seems like a chip reset is required to get them to re-enumerate.

pavel-kirienko, 2015-11-19:
Node ID is not supposed to be affected by enumeration. Do you mean ESC index parameters?

pkocmoud, 2015-11-19:
Pavel, What I see is after enumeration, I get an ID of 125 (1st ESC) or 124 (2nd ESC). Is that the Node ID or ESC Index?

pavel-kirienko, 2015-11-19:
This is node ID.

pkocmoud, 2015-11-19:
It looks to me like PX4 - (Plane mode) may only output throttle commands to ESC on Node 125? I can erase the ESC and re-flash to re-enumerate at NODE 125. There needs to be a way to reset this maybe? Otherwise I might just be confused :)

pavel-kirienko, 2015-11-19:
Node ID has nothing to do with ESC control. PX4 doesn't even know what node ID values ESC have. Could you please describe your issue in a bit more detailed way?

pkocmoud, 2015-11-19:
If there are 2 ESCs connected to a CANBUS, how does PX4 address a single one?

pavel-kirienko, 2015-11-19:
Using ESC index. Multiple ESC may have the same ESC index, which allows to control them simultaneously (if needed).

LorenzMeier, 2015-11-22:
I'm closing this since it seems to be a non issue. We're not supporting one to multi mappings anyway (this is not a Y cable) but require proper configurations for dual-motor setups.

closed 2015-11-22 by LorenzMeier