PX4/PX4-Autopilot #3228: Temporary firmware file can be flashed to wrong UAVCAN node

LorenzMeier, 2015-11-20:
@bendyer the current upgrade method does not take a corner case into account where the buffer file for the firmware upgrade already exists, but the wrong ESC is connected. Can we force updating this buffer file when the firmware upgrade is triggered so it's linked to the current node on the network?

bendyer, 2015-11-20:
It seems like it should be possible to do that more robustly, but I have not looked into the mechanics of the UAVCAN file cache system in much detail yet. @pavel-kirienko or @davids5, what are your thoughts on generating a random name for each file in the cache rather than using the existing file name? I suppose that would mean we would need to clear the cache on startup, unless the names were deterministic, but it's probably safer.

davids5, 2015-11-20:
I am not sure I understand the problem. We have the c (cache directory) that was needed to shorten the paths. It should have files named ```<device><hash>.bin``` ```device``` is needed because a build of all fw (px4cannode,px4esc,etc) will all have the same git hash. This was noted [here] (https://github.com/mavlink/mavlink/pull/414#issuecomment-121640719) 

The fw update process looks for the file in the FQDN path and will copy it to the c directory if it does not exist. 

Then in memory we have the file cache that is used to speed up the programming by managing the file opens from the c directory. It just maintains open FDs for a file that has been opened. But each read does a seek so multiple nodes can use the same file.

Where and what is causing the problem?

bendyer, 2015-11-20:
I think that would fix the problem, but as far as I can see the device name is not included in the cache directory path:
https://github.com/UAVCAN/libuavcan/blob/master/libuavcan_drivers/posix/include/uavcan_posix/firmware_version_checker.hpp#L246-L260

pavel-kirienko, 2015-11-20:
@bendyer @davids5 Indeed, device name is not included in the short path, and it can't be due to path length restrictions. We could use CRC64 computed from full firmware file name for files in the `c/` directory instead of VCS hash. That should fix the problem, or am I missing something?

bendyer, 2015-11-20:
I started using the CRC64 in place of the VCS hash in my makefile to avoid this issue, so that solution is fine by me.

pavel-kirienko, 2015-11-20:
OK. Unfortunately I don't have hardware with me at the moment, is anybody willing to implement this change?

devbharat, 2015-11-20:
We have a sample of both (svesc and px4esc) if you need testing. Am I correct in assuming that now the file name in the cache directory would be the CRC64 computed from the name containing the device name ? (Is it already like that or is there a PR implementing this change ?). 

So now, one does not have to clear the cache directory on the sd-card and connect which ever ESC without worry about flashing the wrong binary ?

bendyer, 2015-11-20:
The change should happen here: https://github.com/UAVCAN/libuavcan/blob/master/libuavcan_drivers/posix/include/uavcan_posix/firmware_version_checker.hpp#L290-L314

I'd suggest opening the original file and checking AppDescriptor etc, then setting the destination filename and `out_firmware_file_path` to the hex value of `descriptor.image_crc`.

I may be able to sort it out tomorrow if nobody gets to it before then.

davids5, 2015-11-20:
The comment in the code wrong  Please re-read https://github.com/mavlink/mavlink/pull/414#issuecomment-121640719

davids5, 2015-11-20:
use the git hash and prefix it 

@pavel-kirienko  @bendyer 

see @pavel-kirienko post https://github.com/mavlink/mavlink/pull/414#issuecomment-133364450

davids5, 2015-11-20:
This is what the fw tree looks like
[image]

The reason for using the git hash is so a human can look at the file and know where it came from

pavel-kirienko, 2015-11-20:
Do humans really need to know where the cached files came from? I see some
value in appending the name, but this approach is susceptible to exceeding
the path length limit.

davids5, 2015-11-20:
What version are you using?

davids5, 2015-11-21:
@LorenzMeier @pavel-kirienko @bendyer @devbharat 

This PR https://github.com/thiemar/vectorcontrol/pull/15 creates uavcan named files per https://github.com/mavlink/mavlink/pull/414#issuecomment-120088482

LorenzMeier, 2015-11-21:
@davids5 Please close once all related PRs are in and tested.

LorenzMeier, 2015-11-22:
@bendyer This PR will address this: https://github.com/PX4/Firmware/pull/3242

bendyer, 2015-11-22:
The dev guide also needs to be updated. 

I think it'd be cleaner to use the CRC to generate unique 8-character cache filenames, since it avoids imposing additional name length constraints on UAVCAN hardware providers, but it seems like that ship has sailed.

LorenzMeier, 2015-11-22:
That's a good point and I'll re-visit that with David.

closed 2015-12-05 by LorenzMeier