PX4/PX4-Autopilot #3236: Initialize multicopter throttle integral to expected hover throttle

LorenzMeier, 2015-11-20:
@tumbili If we initialise the variable ```thrust_int(2) = -i;```, where i == the positive hover throttle, then the multicopter controller will be able to lift off immediately without the need to spool up from the ground.

I would like to have the systems learn this, but initially I think a parameter would be better. The controller is already applying this logic when switching from manual to altitude hold:
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1252-L1263

tumbili, 2015-11-20:
Can't we just ramp up throttle (relatively fast) and a soon as we start to accelerate we can save this as initial throttle value.

LorenzMeier, 2015-11-20:
Good point. Let's try that in SITL.

closed 2016-01-06 by LorenzMeier