PX4/PX4-Autopilot #3243: work_hrtthread() access to invalid address

LorenzMeier, 2015-11-21:
@erikd @mcharleb I fixed a bunch of race / init / list related issues today (will push in a sec) but this crash remains. Is this something which resolves when we port the remaining bits or is this something we should be debugging?

```
pxh> ASAN:SIGSEGV
=================================================================
==40579==ERROR: AddressSanitizer: SEGV on unknown address 0x7fff9bb61c70 (pc 0x00010c187ccf bp 0x00010faa8e70 sp 0x00010faa8e70 T4)
    #0 0x10c187cce in dq_rem dq_rem.c:68
    #1 0x10c186d1e in work_hrtthread hrt_thread.c:146
    #2 0x10c18532c in entry_adapter(void*) px4_posix_tasks.cpp:85
    #3 0x7fff9bb64059 in _pthread_body (/usr/lib/system/libsystem_pthread.dylib+0x4059)
    #4 0x7fff9bb63fd6 in _pthread_start (/usr/lib/system/libsystem_pthread.dylib+0x3fd6)
    #5 0x7fff9bb613ec in thread_start (/usr/lib/system/libsystem_pthread.dylib+0x13ec)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV dq_rem.c:68 dq_rem
Thread T4 created by T0 here:
    #0 0x10c5ab8d9 in wrap_pthread_create (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/7.0.0/lib/darwin/libclang_rt.asan_osx_dynamic.dylib+0x388d9)
    #1 0x10c184fdf in px4_task_spawn_cmd px4_posix_tasks.cpp:180
    #2 0x10c186c02 in hrt_work_queue_init hrt_thread.c:270
    #3 0x10c184959 in px4::init_once() px4_posix_impl.cpp:71
    #4 0x10c16ee16 in main main.cpp:198
    #5 0x7fff927435c8 in start (/usr/lib/system/libdyld.dylib+0x35c8)
    #6 0x1  (<unknown module>)

==40579==ABORTING
Tools/sitl_run.sh: line 94: 40579 Abort trap: 6           ./mainapp ../../../../${rc_script}_${program}_${model}
```

mcharleb, 2015-11-21:
Looks like the hrt_work queue is trying to dequeue a work item that was
already freed somehow. This will go away when the drivers use
driver_framework vs the hrt_work_queue.

LorenzMeier, 2015-11-25:
Fixed.

closed 2015-11-25 by LorenzMeier