PX4/PX4-Autopilot #3245: Firmware tells GCS which MAV_CMD's are supported

DonLakeFlyer, 2015-11-21:
Need this in order to show correct set of available commands in Mission Editor UI. Likely some sort of new mavlink thing.

dagar, 2016-11-08:
This would be nice. Any ideas?

squilter, 2016-11-08:
Could just add to `MAV_PROTOCOL_CAPABILITY`, no?

dagar, 2016-11-08:
There are nearly 100 MAV_CMD's. http://mavlink.org/messages/common
One option could be grouping them into more general capabilities. Camera, mount, vtol, etc.

DonLakeFlyer, 2016-11-09:
Maybe a list of ids not supported in response to AUTOPILOT_VERSION using a mav 2 extension?

dagar, 2016-11-09:
What were you imagining? A variable length array of MAV_CMDs? I suppose we only have to send it once in response to the GCS requesting the list.

DonLakeFlyer, 2016-11-09:
Yup.

dagar, 2017-04-12:
Did we ever get a way to do this over mavlink?

dagar, 2017-04-28:
I'll implement this in PX4 once there's mavlink support.

closed 2018-02-22 by PX4BuildBot