PX4/PX4-Autopilot #3251: Check why a command of 3000 returns as 184, which oddly is 3000 % 256

LorenzMeier, 2015-11-22:
Somewhere between PX4 and QGC someone uses an uint8 where it should be an uint16.

erikd, 2015-11-23:
`uint16_t` or `uint32_t`? I ask, because I found:

```
static const uint32_t VEHICLE_CMD_DO_VTOL_TRANSITION = 3000;
```

sanderux, 2015-11-23:
I can confirm that mavproxy produces the same result.
@erikd It does seem to be a % 256, when trying to send command 257 it returns command 1

sanderux, 2015-11-23:
I cannot find a uint8 casting or % 256 of the command param.
The only thing i could think of was the casting of the mission_item->command to MAV_CMD enum
https://github.com/PX4/Firmware/blob/afb68cb0abee4034582f93d7f383c287b9c9a109/src/modules/mavlink/mavlink_mission.cpp#L818

Enum defined here:
https://github.com/PX4/Firmware/blob/afb68cb0abee4034582f93d7f383c287b9c9a109/src/modules/navigator/navigation.h#L51

And the command param in the mavlink message is formatted by using that nav_cmd here:
https://github.com/PX4/Firmware/blob/fe4c67a781db8a192229f22e70f08a36d7fcd70f/src/modules/mavlink/mavlink_mission.cpp#L866

although i can't replicate this casting behavior perhaps strong casting the enum would help.

LorenzMeier, 2015-12-02:
Will look into this quickly this morning.

LorenzMeier, 2015-12-02:
@sanderux I reviewed the whole data path and did two adjustments. Would you please be so kind to re-test with master from just now and provide the full error log and all details if it persists?

sanderux, 2015-12-02:
@LorenzMeier will do, thanks.

sanderux, 2015-12-02:
Should the current developer build (http://px4-travis.s3.amazonaws.com/Firmware/master/px4fmu-v2_default.px4) already contain your commit? If so the problem persists and i will create a full bug report.

LorenzMeier, 2015-12-02:
Yes, full log please.

LorenzMeier, 2015-12-06:
@sanderux Put a printf() / warnx() in here: https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L3198

Connect the system console and then work your way through it:
http://dev.px4.io/advanced-system-console.html

Let us know if you have any additional information. This should easy to track down with a couple of printfs().

sanderux, 2015-12-06:
I'll get to it this week. Thanks for the tips.

sanderux, 2015-12-21:
Found it.
https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_mission.cpp#L816
casts the mission_item->command to NAV_CMD enum
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigation.h#L66

184 is the last item in the enum.
Simple solution is to add NAV_CMD_DO_VTOL_TRANSITION to the enum, but i think some more sanity checks need to be done. 

i'll get a PR out this week

LorenzMeier, 2015-12-21:
@sanderux Thanks for finding this! I've pushed a fix in 2041e3f. Now on to adding it to the mission 8). Although I believe the better pattern would be to add support to DO_SET_SPEED and perform the transition automatically based on the desired travel velocity (and distance to the next waypoint with hover time). That would allow to auto-script something.

closed 2015-12-21 by LorenzMeier