PX4/PX4-Autopilot #3252: Add support for VEHICLE_CMD_DO_VTOL_TRANSITION in missions

LorenzMeier, 2015-11-22:
Should have this for VTOLs at some point.

LorenzMeier, 2015-11-22:
Note that #3251 needs to be fixed first.

sanderux, 2015-11-23:
I suggest adding a MIS_FAILS_TRANS or VT_FAILS_TRANS parameter that controls the behavior of a failed transition in mission mode. RTL seems like a reasonable default setting.

sanderux, 2015-12-21:
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigation.h#L66
NAV_CMD_DO_VTOL_TRANSITION needs to be added.

sanderux, 2015-12-21:
@LorenzMeier said:
"Although I believe the better pattern would be to add support to DO_SET_SPEED and perform the transition automatically based on the desired travel velocity (and distance to the next waypoint with hover time). That would allow to auto-script something."

I think this would be a good option for tiltrotors, tiltwings and tailsitters. But for standard VTOL i think it would be problematic. When for example a DO_SET_SPEED of 5m/s is set and a transition speed of 10m/s the vehicle would tilt forward (nose down). If after that a DO_SET_SPEED of 15m/s is given the vehicle would transition with it's nose still down. Not a desirable situation i think.

I do think that LAND (at least in failsafe) should transition back to MC as the home position might not be suitable for FW landings.

LorenzMeier, 2015-12-21:
I don't see the argument with nose down. Its up to the transition controller to get that right, and its always a consideration for any transition (be it manually triggered or autonomous). Transitioning depends on desired speed and expected distance to cover and is something that the vehicle can do significantly better than the operator. You're over-burdening the operator during flight planning by assuming he will pre-plan the velocity profile in his head.

It means you're assuming a very technical operator, which spoils the whole concept of autonomous flight to a significant degree.

sanderux, 2015-12-21:
Thats also true. The vtol mixer could level the vehicle before transitioning and the navigator could decide to transition if speed and distance meet the criteria. Is there already a distance estimator for an entire mission? Or should it only be based on the next waypoint? 

i would suggest entire mission as that would give the operator the option to direct the plane into the wind to transition.

LorenzMeier, 2015-12-21:
[image]

I quickly sketched a 4D concept for a mission. The integral defines when to transition.

sanderux, 2015-12-21:
I must be misreading your sketch, i don't see any transition?

LorenzMeier, 2015-12-21:
What I forgot to draw is the minimum transition profile and the transition cost: The minimum transition profile is the minimum distance you need to accelerate, transition, fly, transition back, decelerate. That is (modulo wind) a known fixed quantity of N seconds / N meters. The second metric is transition cost: The whole action likely has an energy premium to it compared to a multi rotor mode hover at slower speed. When both metrics are met the system would transition forward. The back transition is way easier: If the system sees the need to get below a certain minimum speed, it has to transition back.

sanderux, 2015-12-21:
Could the cost factor not be a part of the airframe's params?
MIS_VT_TRANS_SPEED=10m/s // minimum airspeed to consider transition 
MIS_VT_MIN_DISTANCE=500m // minimum distance at speeds > transition speed to consider transition
MIS_VT_MIN_TRANS_DIST=20m // minimum distance to next waypoint to consider transitioning

sanderux, 2015-12-21:
and perhaps;
MIS_VT_MC_TRANS_DIST=10m // distance required to transition back to MC 
this one in case of only a LAND instruction at the end of a long high speed stretch, the navigator could look ahead and anticipate the land by transitioning back a few meters before the waypoint is reached and proceeding in MC mode for the final meters before landing.

sanderux, 2015-12-21:
The later might be better expressed in seconds as this would of course depend on speed

DonLakeFlyer, 2015-12-21:
Just FYI for QGC support: QGC now tracks which mission commands are supported by firmware type. So if new commands are added to PX4 Stack you need to notify me so I can update the list in QGC and have them show up.

LorenzMeier, 2016-01-11:
Done. I have renamed and moved this to a next release.

LorenzMeier, 2016-02-18:
We do it explicitly right now. removing it.

closed 2016-02-18 by LorenzMeier