PX4/PX4-Autopilot #3254: Multicopter mode switch: Review controller state

LorenzMeier, 2015-11-23:
It seems like multicopter mode switches can lead to wild swings in the attitude setpoint, potentially causing NaN values in the attitude controller. We have to understand where these come from.

LorenzMeier, 2015-11-23:
@tumbili FYI

tumbili, 2015-11-23:
At a first glance it looks like the desired acceleration in the mc pos controller is initialized to a large value causing the extreme attitude setpoint. Part of the calculation is a derivative which is suspicious.
Both logs show this behaviour so I hope we can easily reproduce this.

scott-eddy, 2015-11-23:
This [diagram](https://github.com/PX4/Devguide/blob/master/book/images/diagrams/commander-flow-diagram.png) might be help the discussion.  The diagram was put in the devguide with the addition of Rattitude mode.

tumbili, 2015-11-24:
Sorry, my comment was meant for another issue.

LorenzMeier, 2015-11-25:
Fixed.

closed 2015-11-25 by LorenzMeier