PX4/PX4-Autopilot #3255: Bad initialisation of POSCTRL

tumbili, 2015-11-24:
Description to come.

tumbili, 2015-11-30:
User reported this:
I'm a PhD student at McGill, where a few labmates and I have been
assembling a hexacopter based on Pixhawk. We bought the autopilot,
a GPS/compass and a radio pair from 3DR, a Tarot 680Pro frame, and
the (unfortunately) cheap jDrone 30A ESCs and 850KV motors.
We powered this assembly using two 3S 6000mAh batteries in parallel.

We set up the firmware using QGC, and we did the calibration of the
sensors and the ESCs, which I think went well. 

Today we performed a couple of really good flights in which we switched from
manual mode to position control, and we were very impressed at the
stability of the position control mode. We did notice a slight "twitch"
during the transition in terms of pitch and roll, but it handled the oscillation
well.

On the third flight, however, we had a crash that we haven't been able to explain
as a mechanical failure. When our pilot switched from manual mode to
position control, pixhawk basically set the setpoint pitch and roll to what seems
to us as outliers. As a result, the hex rolled almost 180 degrees and fell
of the sky.

We were hoping if you would be willing to help us figure out why this happened.
Is it a matter of PID tuning? If so, how come the transition in one flight is super
stable and in the next it is not?

I'm attaching two data logs from the pixhawk's sd card, one for the stable flight and
one for the one that crashed. Also, a few plots showing the effect of switching
from manual to position control for one of the stable flights vs the one that crashed.

Any advice you have will be useful to us.

LorenzMeier, 2016-01-28:
Assuming this to be fixed as part of the massive rework of that code path since.

closed 2016-01-28 by LorenzMeier