PX4/PX4-Autopilot #3257: Add support for VEHICLE_CMD_DO_SET_MODE for sub-modes of auto

LorenzMeier, 2015-11-24:
Trivial, just requires further switch cases here:
https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L566

NaterGator, 2015-11-24:
I've actually done this already in a personal branch. In addition the [mavlink receiver](https://github.com/PX4/Firmware/blob/483cb1115711d252cb8e7b3bf06acf06734fce9c/src/modules/mavlink/mavlink_receiver.cpp#L540) needs to place the [sub mode](https://github.com/PX4/Firmware/blob/afb68cb0abee4034582f93d7f383c287b9c9a109/src/modules/commander/px4_custom_mode.h#L70) in the command to decipher which auto mode was desired. 

I'll get it in an up to date master and submit a PR.

tumbili, 2015-12-04:
Fixed

closed 2015-12-04 by tumbili