PX4/PX4-Autopilot #3261: FW attitude controller incorrectly based on true airspeed

NaterGator, 2015-11-24:
We have been doing some code review of the fixedwing code and came across a potential issue we wanted to get some eyes on. 

Currently the fixedwing ekf attitude estimator [uses true airspeed](https://github.com/PX4/Firmware/blob/483cb1115711d252cb8e7b3bf06acf06734fce9c/src/modules/ekf_att_pos_estimator/ekf_att_pos_estimator_main.cpp#L902) to set the control state airspeed. From there the fixedwing attitude controller uses the control state airspeed [to define an airspeed scaling factor](https://github.com/PX4/Firmware/blob/483cb1115711d252cb8e7b3bf06acf06734fce9c/src/modules/fw_att_control/fw_att_control_main.cpp#L900) and [sets the ECL control input](https://github.com/PX4/Firmware/blob/483cb1115711d252cb8e7b3bf06acf06734fce9c/src/modules/fw_att_control/fw_att_control_main.cpp#L1068) using true airspeed. 

It would seem appropriate to target indicated airspeed (as well as scale actuation based on indicated airspeed) in the fixedwing controller. @LorenzMeier @tumbili @AndreasAntener I hope you don't mind me tagging you for review...

LorenzMeier, 2015-11-25:
Yes, you're right and we should adjust that. @yyassa we should also fix this for the EKF. This will make the airspeed reading smoother. Also note that we reverted the airspeed change you had in for the EKF since it lagged like hell. Either you have EKF patches locally which fix all this, or you must be getting incorrect data.

NaterGator, 2015-11-25:
In addition to control state publishing IAS, there is also a concern that TECS may inappropriately estimate the system energy due to [the IAS to TAS conversion](https://github.com/PX4/Firmware/blob/ab95b4706e95adac2e64067a908f77b4f4072b5a/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L1109) being disabled.

The position controller used to have the necessary information to generate the EAS2TAS factor correctly but the more concise control state topic doesn't seem to have quite enough info. Not sure the desired direction to fix in the pos controller.

tumbili, 2016-02-17:
@NaterGator @LorenzMeier Wouldn't it make more sense to publish the true airspeed in the control state topic. Then all the control modules don't need to care about conversion and they would all use the same airspeed value.

tumbili, 2016-02-17:
Hm, not for the attitude controller though, I see the point.

LorenzMeier, 2016-02-18:
Fixed a long time ago.

closed 2016-02-18 by LorenzMeier