PX4/PX4-Autopilot #3264: Control state broke vtol airspeed

tumbili, 2015-11-25:
The VTOL stack has been using INAV as a default until now. The problem is that the attitude estimator_q which does not write airspeed data into the control state topic which at the moment makes it unusable for fixed wing operation.
The question (until we get the new ekf in and running) is if we want to switch to using the ekf for VTOL or if we want the attitude estimator q to subscribe to airspeed and put the value into the control state topic?
@LorenzMeier @yyassa

yyassa, 2015-11-25:
What about channeling the raw pitot measurement over inav and publishing it from there as ctrl_state.airspeed? (as the inav is not estimating wind)

tumbili, 2015-11-25:
@yyassa I think that's what we're going to do but we need to do it in the q estimator

yyassa, 2015-11-25:
Sure, I can do the fix quickly.

General question: As far as I understand you cannot publish  part of a topic from one module and the other part from a different one?

LorenzMeier, 2015-11-25:
No, you can't.

closed 2015-11-30 by tumbili