PX4/PX4-Autopilot #3270: Sudden magnetic declination change screws up yaw estimation

muharred, 2015-11-25:
On init and in cases where magnetic declination changes considerably in one leap attitude estimation by yaw slowly reacts to the change and during the time accumulates false gyro bias (because mag angle adds declination instantly, but gyro doesn't register any rotation). This bias, in turn causes yaw estimation to overshoot the new level and can cause unstable yaw for more than a minute after init.
Steps to reproduce: boot with instant logging enabled and with noticeable mag declination stored in ATT_MAG_DECL parameter. Wait 1.5 minute, modify the parameter or start fake GPS. Notice slow yaw reaction with overshoot.

Log file attached.
Screenshot comparing raw mag angle (with NED projection) and yaw estimation (declination was changed from 6.0979 to -6.0979 and back and then fake GPS was turned on):

[image]

LorenzMeier, 2015-11-25:
Thanks for reporting! The fix will be to reset the filter state to the current roll, pitch and heading (based on the mag) once the declination parameter changes.

muharred, 2015-11-25:
I've added pull request with proposed fix that seemed to do the trick for me.

tumbili, 2015-12-04:
Closing, fixed in #3271

closed 2015-12-04 by tumbili