PX4/PX4-Autopilot #3279: SITL timing

LorenzMeier, 2015-11-27:
The recent changes to SITL introduced a timing sensitivity which in my tests always happened exactly after 50 seconds on my machine. First the timing goes bad (the simulation rate drops considerably), then the controller become unstable. There are several things needing inspection / work:

[ ] The recent changes introducing the effect need review
[ ] We anyway want to move to a lockstep architecture
[ ] We need to ensure there is no buffering on the transport side (UDP / protobuf)

[image]

erikd, 2015-11-27:
Steps to reproduce?

LorenzMeier, 2015-11-27:
To repro:
```make posix jmavsim```

Then in the PX4 terminal after home is set:
```commander takeoff```

The terminal will indicate where the log file is put. This folder will be in build_posix_xxx/src/firmware/.. The log file can be read with flight plot: https://pixhawk.org/dev/flightplot

Or uploaded to http://logs.uaventure.com.

LorenzMeier, 2015-11-29:
@erikd @mcharleb I'm wondering if this is related to to us using two separate methods for IPC between threads right now. It has all symptoms of bad scheduling. The SITL executable runs at 250% CPU on my machine suggesting one thread is in a tight loop.

LorenzMeier, 2015-11-29:
@erikd @mcharleb I've narrowed this down to the baro sim. Something is not quite right, it seems to go into a tight loop.

erikd, 2015-11-29:
I've reproduced the problem here. Couple of interesting points:

* Over the weekend, testing on my Linux laptop it was never stable for more than a couple of seconds.
* On the first run this morning on my desktop Linux system, it behaved as you described, stable for 30 seconds or more before going wildly unstabled.
* On subsequent runs, it was stable for about 5 seconds or less.

That's when I realised that in order to prevent the px4 simulator messing up my terminal, I was killing it with Ctrl-C instead of issuing a `shutdown` command. I noticed that when when the app was killed with Ctrl-C, the following run was likely to become unstable almost immediately, while if terminated with the `shutdown` command, the following run would be stable for 30 seconds or more.

Another thing I have noticed is that if I run `make posix jmavsim` and then just leave ir running with no further interaction, the application will segfault after about 4 or 5 minutes. I've tried running this under valgrind but under valgrind it doesn't crash. Running under GDB is no good because I can't convince GDB to ignore `SIGCONT`. One other thing I notice is that when running under valgrind, I get a periodic `WARN  GPS timeout` message which does not happen whn running without valgrind.

I will take a look at the baro sim.

LorenzMeier, 2015-12-02:
I'm closing this for now, as Lockstep is a feature request, not an issue.

priseborough, 2017-06-18:
This should be re-opened. It is a major source of estimator failures in simulation and is a blocker to the use of simulation to for integration of vision sensing data.

tumbili, 2017-06-19:
+1 For reopening this. I've been trying to use SITL as a development tool for the estimator but I've given up due to inconsistency in the data. At this point I don't think that this issue is just related to computers with less computational resources, my laptop is pretty good but I still have these issues.

priseborough, 2017-06-19:
I agree. Here is a plot of the integrated sensor_combined inertial nav delta time and the system time from a gazebo simulation on a mac powerbook which has plenty of power:

[image]

I get the same result when running jmavsim.

tumbili, 2017-06-19:
@priseborough If I'm not mistaken the delta time for the sensor combined topic is calculated by taking the difference of the timestamps which stems from system time. So I don't quite understand where this difference comes from. How were you measuring the system time?

TSC21, 2017-06-19:
Thumbs up fot reopening this. I have been facing strange issues which I believe may come from this.

priseborough, 2017-06-19:
@tumbili the sensor module uses the integral_dt value from the imu sensor report, not the system time delta.

My plot of system time came from the log timetamp.

tumbili, 2017-06-19:
@priseborough True, but the integral dt value from the sensor topic stems for timestamps which are calculated using the system time. See e.g. the timestamp for the gyro topic in SITL
https://github.com/PX4/Firmware/blob/master/src/platforms/posix/drivers/gyrosim/gyrosim.cpp#L1025
So my question still is why do we see those differences in timing.

bkueng, 2017-06-19:
@tumbili it could also be that the logger misses some topic updates due to inaccurate scheduling. The same can happen for the estimator.

closed 2018-02-22 by PX4BuildBot