PX4/PX4-Autopilot #3283: Oscillating in manual control

BrownKyle, 2015-11-27:
I am attempting to test Mocap support #2317, with Mavros. It's was recommended that the master build was used, however when my quadrotor has the master build of pixhawk px4 firmware it oscillates in manual flight. This wasn't an issue with the stable build.

My logs are:
http://logs.uaventure.com/view/Ujvh3pm2JzNQeX5HLDp7sC
http://logs.uaventure.com/logmuncher/charts/Ujvh3pm2JzNQeX5HLDp7sC/logmuncher.log

Any help is appreciated.

LorenzMeier, 2015-11-27:
@tumbili We need to check if the default output rate is still set to 400 Hz for all platforms.

LorenzMeier, 2015-11-28:
Could you provide more details on your setup? Which board? Where did you connect your motors?

BrownKyle, 2015-11-28:
I used a 3DR pixhawk, the motors were connected to the servo ports 1,2,3 and 4.

LorenzMeier, 2015-11-28:
MAIN or AUX port?

BrownKyle, 2015-11-28:
MAIN

LorenzMeier, 2015-11-28:
Ok. Do you have a log file from stable as well?

BrownKyle, 2015-11-28:
No I dont, I wasn't sure which log was the most recent so I deleted them all prior getting the log posted.
I can flash the stable and fly, but not until Monday as the multi rotor is in my university.

BrownKyle, 2015-11-30:
Hey, @LorenzMeier I performed the flights. The flight was stable on the stable release. Here are the logs.

Plots: http://logs.uaventure.com/view/4W77E3s8fbWd6uc5TGbG5n
Log: http://logs.uaventure.com/logmuncher/charts/4W77E3s8fbWd6uc5TGbG5n/logmuncher.log

LorenzMeier, 2015-11-30:
Ok, two main things:

  * You might want to try master from today. The phase margin improved.
  * Your log shows that your system was previously already marginal and not well tuned. See the huge angle offset: http://logs.uaventure.com/view/4W77E3s8fbWd6uc5TGbG5n#AS_Roll_PLOT and also note the rate offset: http://logs.uaventure.com/view/4W77E3s8fbWd6uc5TGbG5n#A_RollRate_PLOT

I would advise to balance your airframe, increate the MC_ROLLRATE_I and MC_PITCHRATE_I gains and re-try with master.

BrownKyle, 2015-11-30:
It works now on master. Thanks, still some issues but its flyable so I will fox the gains.

tumbili, 2015-11-30:
@BrownKyle Could you upload the log with latest master where you used the same gains like you used in stable?

tumbili, 2015-11-30:
The words "still some issues" makes me think that there is still something different between stable and master.

BrownKyle, 2015-11-30:
I have flights with the master working, but I started changing the gains shortly after. I have the log which I thick is the one from after flying with the most recent master. I can't be sure, is there a way to get the filename of the log from log muncher?

Since the flight I have been changing the gains and have just started going through the tuning process detailed on pixhawk.org. Therefore my gains are no longer the same, is it possible to get the aircraft parameters from log muncher and I can redo the flight? If not I could reset the frame to generic quad it was initially set to and then fly again?

BrownKyle, 2015-11-30:
Here are the logs.

Plots: http://logs.uaventure.com/view/kAuwLAYAdRUaVgAzKZW9Pn
Log: http://logs.uaventure.com/logmuncher/charts/kAuwLAYAdRUaVgAzKZW9Pn/logmuncher.log

I increased the pitch rate and roll rate integral gains from 0.05 to 0.055. But the gains are the same except from that. The flight was a little twitchy. The odd thing was it would fly twitchy and then very smooth, it wasn't too consistent.

BrownKyle, 2015-11-30:
I did a flight with the Generic Quad, gains. they appear to be different from the initial Generic Quad gains I had.

The flight seemed good at first, pitch and rolling seemed fine, except when you do both. The quad became unstable and difficult to land.

Plots: http://logs.uaventure.com/view/bFmmZDPriYkTZ9oW3rLCr3
Log: http://logs.uaventure.com/logmuncher/charts/bFmmZDPriYkTZ9oW3rLCr3/logmuncher.log

I will set up the quad will the original gains and test that.

BrownKyle, 2015-11-30:
So I changed the gains to the original set. It wasn't different from the generic quad on the current master, just the acro limits where different, which I don't imagine will have an effect as we were flying in manual/angle mode.

The logs are
Plots: http://logs.uaventure.com/view/jRN9Z8EzXVWcpPMDXhXbiP
Log: http://logs.uaventure.com/logmuncher/charts/jRN9Z8EzXVWcpPMDXhXbiP/logmuncher.log

The system seems to be very stable and without control inputs other than throttle will hoover in place very nicely. But there is still some small oscillation and it seems to be random, but no where near as bad as it was originally.

tumbili, 2015-11-30:
Ok, I checked the log and it looks very similar to logs from stable. I guess we can close this then.
Feel free  to reopen if you feel like there is still a problem with this.

closed 2015-11-30 by tumbili