PX4/PX4-Autopilot #3286: Check multicopter navigation logic

LorenzMeier, 2015-11-28:
From here:

https://groups.google.com/forum/#!topic/px4users/5vYicnpcmWo

Is there an error in mc_pos_control_main.cpp?

https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L993

```C
 /* copter is too far from trajectory */
						/* if copter is behind prev waypoint, go directly to prev waypoint */
						if ((pos_sp_s - prev_sp_s) * prev_curr_s < 0.0f) {
							pos_sp_s = prev_sp_s;
						}

						/* if copter is in front of curr waypoint, go directly to curr waypoint */
						if ((pos_sp_s - curr_sp_s) * prev_curr_s > 0.0f) {
							pos_sp_s = curr_sp_s;
						}
```

In the code , the (pos_sp_s - prev_sp_s) is equal with prev_curr_s  , so the programme while never less than 0.0f.

Is this?

tumbili, 2015-11-28:
I don't see a problem here. The vector (pos_sp_s - prev_sp_s) will always be parallel to prev_curr_s but it's direction can be opposed in which case the dot product of the two vectors will be negative.

LorenzMeier, 2015-11-28:
Indeed, closing.

closed 2015-11-28 by LorenzMeier

CornerOfSkyline, 2015-11-29:
In this line ：https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L930
		math::Vector<3> pos_sp_s = curr_sp_s;
And in this line:
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L940
				math::Vector<3> prev_curr_s = curr_sp_s - prev_sp_s;

So, (pos_sp_s - prev_sp_s) * prev_curr_s = (curr_sp_s - prev_sp_s) * prev_curr_s  = prev_curr_s   * prev_curr_s   = 0 . 
and
If you want to find where the copter is,I think  the "pos_sp_s"  should be "pos_s“ , is it right?

@tumbili 
@LorenzMeier