PX4/PX4-Autopilot #3291: Mavlink rate control doesn't allow high stream rates.

mhkabir, 2015-11-30:
I'm trying to get IMU data out over mavlink at 250Hz, which should be easy, But the rate limiter caps the data rate to 174 Hz.
Mavlink is running in "onboard" mode, over TELEM2 at 921600 baud.

This is easily reproducible by connecting over TELEM2 to e.g QGC and inspecting the default "ATTITUDE" message which is supposed to be streamed at 200Hz in onboard mode.

Note that the serial port on my board has no hardware flow control pins.
Help would be appreciated!

mhkabir, 2015-11-30:
When only running the TELEM2 link : 
```
instance #0:
        mavlink chan: #0
        no telem status.
        rates:
        tx: 20.411 kB/s
        txerr: 0.000 kB/s
        rx: 0.000 kB/s
        rate mult: 1.000
```
So it seems that rate limiting shouldn't be affecting things, since the multiplier is at 1.0.

Shokman, 2016-02-15:
We are having the same issue, do you have any idea why is this?

mhkabir, 2016-02-16:
Yeah. I believe that increasing the ```-r``` parameter on the companion link had worked, in order to get higher data-rates.
https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L515

LorenzMeier, 2016-05-22:
You need to manage the max data rate param.

closed 2016-05-22 by LorenzMeier