PX4/PX4-Autopilot #3295: Local frame reference shouldn't change in flight

AndreasAntener, 2015-11-30:
I recently started to observe offsets when using position control relative to my takeoff location, using local frame coordinates. INAV resets the local frame reference when the home position updates. This didn't affect any of our offboard code until recently, I believe the reason for it is the new (maybe it's already ancient) home position update on takeoff. Depending on the time that passes between arming and takeoff this just gives a slight but noticeable offset.

As far as I understand EKF, it only updates the local reference only once in the beginning. But also when a filter reset occurs which could be in-flight.

This is really problematic when using local position setpoints. What would speak against locking the local frame reference for estimators once determined with good GPS reception?

LorenzMeier, 2015-11-30:
INAV should indeed not reset it ever.

AndreasAntener, 2015-11-30:
Ok, INAV already initializes the ref when GPS is good, so removing the home update logic entirely  seems to be a straight forward fix. Will do that and include in what we test the next days.

LorenzMeier, 2015-12-31:
Fixed.

closed 2015-12-31 by LorenzMeier