PX4/PX4-Autopilot #3296: Unexpected behaviour in calling Matrix inversed()

icoderaven, 2015-11-30:
I just noticed that calling an inversed() set's the class object data to identity
i.e. _params.inertia_inv = _params.inertia.inversed();
where inertia_inv and inertia are 3x3 matrices sets the desired inertia_inv matrix to the correct value, but resets the data of the inertia matrix to identity (maybe as a by product of the decomposition) This seems to suggest that the inverse is obtained in-place as opposed to on a copy.

LorenzMeier, 2016-06-21:
@tumbili Would be great to close this.

tumbili, 2016-06-22:
@icoderaven @LorenzMeier I cannot replicate this. See my code below:
```
math::Matrix<3,3> A;
		A(0,0) = 2;
		A(1,1) = 1
		A(2,2) = 5;

		math::Matrix<3,3> B = A.inversed();

		A.print();
		B.print();
```

Result:
```
[ 2.000	0.000	0.000	 ]
[ 0.000	1.000	0.000	 ]
[ 0.000	0.000	5.000	 ]
[ 0.500	0.000	0.000	 ]
[ 0.000	1.000	0.000	 ]
[ 0.000	0.000	0.200	 ]

```

@icoderaven Could you verify if this problem still exists and if yes, tell us how to replicate it?

closed 2016-06-22 by tumbili