PX4/PX4-Autopilot #3299: OFFBOARD control(setpoint local position) not working

jabberwack, 2015-12-01:
I just updated my firmware of my pixhawk to the newest firmware through qGroundcontrol.
Afterwards, the offboard control with setpoint local position stopped working.
I checked with my old quad which had the old firmware, and the exact same code worked fine on it.
Could you guys check if something went wrong with you updates?

Thanks guys.

Aperture-Laboratories, 2015-12-01:
Are you on master? 
What's your connection to the pixhawk like on both quads for offboard control? 
Did you update anything else? Mavros just got some toppic name updates that "broke" the code I was running before updates.

BrownKyle, 2015-12-01:
I had a similar issue with the latest master. I had data from my motion capture system streaming to the pixhawk, through ROS I was able to switch the system to off board control on a relatively recent master. On the very latest master the flight controller doesn't seem to be using the motion capture data into its position estimate. I have not tried off board control since, as the position data wasn't working. I thought it might be my ROS install was messed up from a series of modifications and was working on reinstalling everything. Thought I would mention it when I saw this issue.

I never actually flew the multi rotor in off board mode in the previous master, as the multi rotor was very unstable until a recent update.

jabberwack, 2015-12-02:
Yes, I was on master(about a week ago).
I have connected my pixhawk and my offboard with USB to UART module which is plugged in to the telem2 port of pixhawk. Rate set to 921600.
Checking the log, Pixhawk seems to receive the setpoint local position coordinates correctly 
(LPSP_X, LPSP_Y, LPSP_Z were all received).

Thanks.

jabberwack, 2015-12-04:
I checked by installing the previously released firmwares and the offboard control  for setpoint localposition stops working after the second release of 1.0.0 beta.
First release of 1.0.0 beta works fine.

jabberwack, 2015-12-08:
nevermind.
updating mavros solved the problem!

closed 2015-12-08 by jabberwack