PX4/PX4-Autopilot #3305: Protect against missing airspeed

LorenzMeier, 2015-12-01:
http://logs.uaventure.com/view/nstjSNj7LvizLj8m3CDdeA

tumbili, 2015-12-02:
Control state needs to consider non updating airspeed sensor.

LorenzMeier, 2015-12-04:
@tumbili From your closed PR: you should do this:

```diff

-		/* Airspeed - take airspeed measurement directly here as no wind is estimated */
-		ctrl_state.airspeed = _airspeed.indicated_airspeed_m_s;
+		// Airspeed - take airspeed measurement directly here as no wind is estimated
+		if (PX4_ISFINITE(_airspeed.indicated_airspeed_m_s) && hrt_elapsed_time(&_airspeed.timestamp) < 1e6) {
+			ctrl_state.airspeed = _airspeed.indicated_airspeed_m_s;
+			ctrl_state.airspeed_valid = true;
+
+		} else {
+			ctrl_state.airspeed = 0;
+			ctrl_state.airspeed_valid = false;
+		}
```

We need the valid airspeed range detection still somewhere. I would propose to start with airspeed < ground_speed / 2 or something similarly simple.

LorenzMeier, 2015-12-04:
Of course this also needs the controller to copy that flag.

tumbili, 2015-12-07:
@LorenzMeier Yes of course, will fix this up today.

LorenzMeier, 2015-12-07:
@tumbili This one is the last issue for 1.1.0 without a proposed PR now.

closed 2016-02-18 by LorenzMeier