PX4/PX4-Autopilot #3309: Differentiate between (override) RC and joystick

LorenzMeier, 2015-12-02:
So far we considered the joystick and the RC the same device via different physical links. It seems like it would be good to differentiate a bit more:

  * Create a msg/joystick_input.msg file for MAVLink joystick commands
  * Store the last RC control state. If joystick data comes in and the RC is unchanged, use it.
  * If the RC control state changes and is in manual, lock out the joystick data (this is intended for override)
  * If the RC control state is non-manual and joystick stops, enable RC. This allows to fly home in position control when exiting the app.

@DonLakeFlyer Did I cover all corner cases? It sounds a bit ugly, but the best I could come up with.

tubeme, 2015-12-19:
Very good way of functioning. Probably good idea for the pilot and his team is to have a way to manually switch it back and forth from QGC. Vasil Petrov

DonLakeFlyer, 2016-02-29:
Should there be a way to force control to one side of the equation? Force joystick, or force radio? Default would be both. But maybe at some point someone wants to take over control. Not sure. Will also need some sort of deadband filtering for jitter.

I don't quite get why you need to distinguish things by fllight mode?

superware, 2018-02-11:
Hi @LorenzMeier, is this currently implemented? or is there still a conflict between RC and joystick commanding?

Stifael, 2018-08-30:
https://github.com/PX4/Firmware/pull/10227 goes in a similar direction

AuterionWrikeBot, 2019-01-17:
➤ Lorenz Meier commented:

@Beat Küng@Julian Oes This is still an issue.

beat, 2019-01-17:
Please stop using my github id for Beat Küng. He has @bkueng as id. Thanks! :wink:

LorenzMeier, 2019-07-09:
I'm closing this for now as we're overall reworking the inputs and the current way of doing things seems to generally work - RC is quickly becoming a legacy thing and not connected in parallel.

BazookaJoe1900, 2019-07-10:
@LorenzMeier @dagar 
I don't agree that RC become legacy. And really waited to this feature. 
There are few scenarios: 
-	If the system is stable, and became full autonomous it doesn't need RC. 
-	For test flight we get many drones saved because the auto modes had problems and we take control to RC.
-	On Fully deployed system, where there is no operator that know how to Fly RC. Or the system is beyond of the RC radio signal. The operator might want to take a joystick control.
-	In case of loss of GPS, the operator can take control to joystick attitude mode and fly in FPV mode.

LorenzMeier, 2019-07-10:
@BazookaJoe1900 That's all fine, but will you work on it? Otherwise I'm going to keep it closed because for us it is a "wontfix" issue. There are lots of good suggestions for PX4, but we need to prioritize the work of the core dev team and not have issues dangling around that we have no plan to address.

We need to get better at setting expectations, and part of that is saying no to things. If you are willing to go ahead and contribute the implementation for this in the next weeks then we can make sure to support you on it.

closed 2019-07-10 by LorenzMeier