PX4/PX4-Autopilot #3310: Support libm in IO config

LorenzMeier, 2015-12-03:
@davids5 We need to put sqrt() in there, compile and fix any resulting compilation fails:

In the mixer_multitotor.cpp using tan() and tanf() has no error, but using sqrt() or sqrtf() cause the link error: undefined reference to '__errno' when px4fmu building.

davids5, 2015-12-03:
No problem found on Firmware @ master for px4fmu-v2_default with  sqrt() added. 
building with:
```
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.8.4 20140725 (release) [ARM/embedded-4_8-branch revision 213147]
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

LorenzMeier, 2015-12-18:
Already fixed in master. Was a stable issue.

closed 2015-12-18 by LorenzMeier