PX4/PX4-Autopilot #3316: Pixhawk inconsistent boot-up behavior with master and beta

aharmat, 2015-12-03:
We're running pixhawk with PX4, using stable firmware the boot-up behaves as expected. With both master and beta firmware, one of three things happens:

1. Boot-up behaves as normal
2. Pixhawk boots normally (based on noise), one or more ESCs keep beeping with motors twitching
3. Pixhawk emits a continuous tone, one or more ESCs keep beeping with motors twitching.

There is no apparent pattern to these behaviors, cycling the power eventually gets us to normal boot-up. The quad flies normally after that. Any idea what's wrong?

Thanks,

Adam

LorenzMeier, 2015-12-04:
We're not aware of any recent issues - what vehicle configuration are you using? Are you flashing via QGroundControl?

aharmat, 2015-12-04:
We are using a hexacopter + configuration, flashing via QGroundControl. Does this sound like an ESC calibration issue?

LorenzMeier, 2016-05-22:
Outdated.

closed 2016-05-22 by LorenzMeier