PX4/PX4-Autopilot #3320: jMAVSIM: Unstable with SITL

tumbili, 2015-12-04:
@mcharleb 
I still have the issue that SITL becomes unstable when using jMAVSim.
I can fly for about 1 - 2 minutes but then it starts oscillating and crashes. During this the cpu load increases massively on all cores. I've tried to stop logging and reduced the mavlink streams to the bare minimum but the problem also occurs then.

I've also conducted some experiments in the simulator module:
I've measured the time differences between the imu messages received from jmavsim.
When the simulation is stable I get a delta t of around 4ms. As soon as  the simulation gets unstable I see values like 0.1 ms which is not really realistic anymore.

How can this be explained? Could it be that the hrt_timer does not work properly anymore because there is a busy loop somewhere?

LorenzMeier, 2015-12-04:
When this happens next time run the "perf" command and copy the output here.

LorenzMeier, 2016-01-06:
Fixed.

closed 2016-01-06 by LorenzMeier

tumbili, 2016-01-07:
@LorenzMeier What was the problem?

LorenzMeier, 2016-01-07:
PWM min and max values. The need for lockstep remains.