PX4/PX4-Autopilot #3324: Battery voltage not available with latest master

AndreasAntener, 2015-12-04:
From what I can see via MAVLink I'm guessing that this flag is never set to true (condition_battery_voltage_valid): https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L1812

However, we were able to arm and fly the quad, not sure if this should be possible. The commander only locks arming down if the above mentioned flag is set, otherwise no battery check is performed (except avionics on arming): https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L1886

Koensw, 2015-12-04:
We also got this problem, after reverting to a previous version of sensors.cpp it was resolved. I just looked into the code, not 100% sure but this line seems to be a problem https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensors.cpp#L1622, _fd_adc is not initialized anymore after https://github.com/PX4/Firmware/commit/d2cacb9bc65ef1bdf10177ac11ddef609f3ce55.

LorenzMeier, 2015-12-04:
*sigh* This slipped through the review. Will fix it tomorrow.

LorenzMeier, 2015-12-04:
Fixed on master.

closed 2015-12-04 by LorenzMeier

AndreasAntener, 2015-12-04:
Great, thanks!