PX4/PX4-Autopilot #3342: Make battery failsafe RTL GUI-configurable

LorenzMeier, 2015-12-08:
We have all the bits in place, but its hidden deep in the failsafe handling config. We need to potentially simplify that bit (to the point of adding a new, simpler non-OBC failsafe mode), make it easily configurable and expose it in QGroundControl.

LorenzMeier, 2015-12-08:
Failsafe should be triggered here:
https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L1903

And we need to do the equivalent of that:
https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L1996-L2078

I would prefer if we could combine that into one block so that it triggers RTL on geofence or low battery and that you can abort that with a mode switch or stick motion.

LorenzMeier, 2016-04-23:
@DonLakeFlyer The param name is COM_LOW_BAT_ACT == 0 is off, 1 is RTL on low battery

LorenzMeier, 2016-04-25:
Finally done!

closed 2016-04-25 by LorenzMeier

blandry, 2016-04-28:
@LorenzMeier we want to handle the case where we don't have GPS but need to land because of a battery failsafe. The logic for that happens in the navigator but you never reach it because in a situation like that your transition to RTL would be flat out refused by the main state transition. I can think of a few fixes, but this seem to point to a potentially deeper problem with the coupling between the FSM's (the main state one is preventing the navigation one to handle the fallback properly).

LorenzMeier, 2016-04-29:
I don't think so. Landing will only be rejected if no position estimate is available. Is has absolutely nothing to do with GPS.

LorenzMeier, 2016-04-29:
@blandry That doesn't mean I'm not interested in feedback for a commander cleanup / rewrite sprint. Its just that the architecture with the navigator perfectly holds up in a GPS-denied world.

blandry, 2016-04-29:
I was referring to this check (status_flags->condition_global_position_valid && status_flags->condition_home_position_valid) which you are right is not GPS per say. My point still stands though, if you don't have a valid position estimate and are in a battery failsafe you probably want to reach vehicle_status_s::NAVIGATION_STATE_DESCEND which you will not reach. In other words in state_machine_helper.cpp line 377 and line 761 disagree as to what to do in that state+status and I don't think they should. Line 761 (navigator version) is right about what to do in the battery case, not 377 (main state version).