PX4/PX4-Autopilot #3345: Inspect altitude hold performance

LorenzMeier, 2015-12-08:
@AndreasAntener @tumbili You have until December 15 to improve the altitude hold performance to a comparable level to the reference you're currently testing. So please plan the time wisely.

@AndreasAntener @SimonWilks December 15 is a hard limit to Roman's availability. Please account that in your scheduling.

blandry, 2016-01-29:
Has there been any progress on this? Flying offboard control (with velocity setpoints) makes the copter loose a dangerous amount of altitude. Forcing _run_alt_control to true seems to fix it, and since a P gain on position should be in I gain velocity I tried that too but didn't get the same level performance. Thoughts? (ps: noticeable in both Gazebo and on a real copter).

AndreasAntener, 2016-01-30:
@blandry Hey Benoit, this issue is basically about general altitude hold performance (from estimator to controller). I started to track down the resent issues/PRs for velocity control: https://github.com/PX4/Firmware/issues?utf8=%E2%9C%93&q=milestone%3A%22Offboard+Velocity+Control%22+
I'm sure it's not far off from working as it should (it did once), just needs someone to get his head around it, and a few more users who regularly test it.

closed 2016-02-18 by LorenzMeier