PX4/PX4-Autopilot #3348: wrong altitude setpoint when switching to altitude control

ChristophTobler, 2015-12-09:
Had a weird behaviour when switching to altitude control.
http://logs.uaventure.com/view/Z9gCEDujtGeJCTEj7dfkd7#Local_Z_PLOT

Must be something with reset_alt_sp and _sp_move_rate(2)

tumbili, 2015-12-09:
I will look into this, seems to be important for the release.

ecmnet, 2016-01-09:
Ok. That is what I found also. Please refer to 

http://logs.uaventure.com/view/sQV6rBwSbQaP3P8vXf2WpV

This happens not all the time, currently it seems to be a random behavior. Although I am new to the PX4 flightstack I am trying to rush through the code.

ecmnet, 2016-01-09:
@tumbili : Did you already had a chance to dig into it? If so, could you share your results here?
Thank you

ecmnet, 2016-01-09:
Maybe another comment: It seems as if the controller takes action, before the set point was reset.

ChristophTobler, 2016-01-09:
I think you can recreate it by doing: arm -> fly with alt hold -> land and disarm -> arm again -> fly and enable alt hold

ecmnet, 2016-01-09:
Hm. I dont be sure. As far as I remember, it occurs also the first time arming...

LorenzMeier, 2016-01-10:
More logs: @tumbili This issue is to track the altitude hold reset logic fails we've seen.
http://logs.uaventure.com/view/Z9gCEDujtGeJCTEj7dfkd7#Local_Z_PLOT

LorenzMeier, 2016-01-10:
@tumbili It would be good if you could have a solid look tomorrow.

ecmnet, 2016-01-15:
As far as I found out, this issue has to do with 'reset_alt_sp' using the FF term at the moment of enabling altitude control. With

	_pos_sp(2) = _pos(2) ;//+ (_vel(2) - _params.vel_ff(2) * _vel_sp(2)) / _params.pos_p(2);

disabling the FF term, the set point is not set accurate, but there seems to be no 'jump' anymore.

tumbili, 2016-01-15:
@ecmnet I will look at it today.

ecmnet, 2016-01-15:
Thank you. Using my modification the result is like this:

http://logs.uaventure.com/view/7t7C7iLc3M6jNKEQWNbwP3#Local_Z_PLOT

tumbili, 2016-01-18:
@ChristophTobler @ecmnet Which branch do I need to have a look at. It seems that this problem is only occurring when using the lidar.

ecmnet, 2016-01-18:
@tumbili I used the master branch, but added Christoph Toblers 'position_estimator_inav' - version
(https://github.com/ChristophTobler/Firmware/commit/9c30fc5027c1c3867e15ac32e295929eda1d5771).
I don't know, whether Christoph had a branch for this.

On the other hand, position estimation with his modification looks fine also in manual mode. I assume, you're investigation goes in the direction, that _vel(2) is not correct, but I haven't seen this.

ecmnet, 2016-01-20:
@ChristophTobler @tumbili Any further results here? As Christophs enhancement of 'position_estimator_nav' is proposed for release 1.2, this should also be fixed.

tumbili, 2016-01-20:
@ecmnet Yes, I know now what the problem is, will push a fix shortly.

tumbili, 2016-01-20:
Fix proposed in #3549

ecmnet, 2016-01-20:
@tumbili Looks good. Did you test it already in flight? I will do that on Friday.

tumbili, 2016-01-20:
@ecmnet No I did not test it yet.

tumbili, 2016-01-22:
@ecmnet Ok, I pushed another commit to #3549. Have a look at it and let me know what you think.
By the way, can you try takeoff on master with the exact same paramters as last time (the ones which leas to broken props?). It should be really nice now!

ecmnet, 2016-01-22:
I haven't tried takeoff yet, as I am currently limited to fly indoors and I fear to crash into the ceiling... ;-)

tumbili, 2016-01-22:
@ecmnet I'll do a video of a takeoff to 1.5 meters, maybe that will convince you ;-)

ecmnet, 2016-01-22:
Well, in return, I'll post a pic of the traces on my living-room ceiling ;-) that my previous java-based flight-stack caused...

LorenzMeier, 2016-03-13:
We fixed this.

closed 2016-03-13 by LorenzMeier