PX4/PX4-Autopilot #3359: Check MC_POS_CONTROL for use of actual vs. set point yaw for transformations

LorenzMeier, 2015-12-14:
@tumbili Please check tomorrow that in mc_pos_control the right yaw value is used for transformations. Andreas has some toilet-bowling which very much sounds like the wrong value is being used.

psiorx, 2015-12-21:
This might also be caused by using the external heading update.  I noticed that when the vehicle is yawed 90 degrees clockwise while using mocap, the px4 and mocap estimates start to diverge which causes toilet-bowling in position control mode.

LorenzMeier, 2015-12-28:
@tumbili The line I'm not 100% about is this one:
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1650-L1651

LorenzMeier, 2015-12-28:
@julianoes Could you have a look here?

LorenzMeier, 2015-12-28:
If it is indeed wrong this here would be the fix: https://github.com/PX4/Firmware/pull/3397

I'm just not 100% convinced yet its wrong. The fix does not deteriorate anything in simulation at least.

LorenzMeier, 2016-01-28:
We checked and proved its correct now.

closed 2016-01-28 by LorenzMeier