PX4/PX4-Autopilot #3362: Add smart battery support

AndreasAntener, 2015-12-15:
driver and integration

myLens, 2016-01-11:
hello andreas,
a created a topic "redundant battery" in the p4users group. could you please explain if your idea of "smart battery" incorporates battery o-ring and  failure detection resulting in a reason for urgent landing (I'm quite new so my 'expressions' might be wrong). currently i use a GPIO of the IO-processor to supervise open drain outputs from all battery o-ring controllers attached and i use a new alarm status bit and a very basic alarm_status.msg for higher layers.
Sure this is not the proper way of doing things and maybe it is even weird to use the 'failsafe-processor' here. what is your suggestion for a solution not waiting for 'smart batteries' that could be everything and nothing in terms of redundant and SPOF avoidance.
Thank you
Tom

AndreasAntener, 2016-01-11:
@myLens with smart battery I'm referring to the work in #3402 and the integration of the data provided by those batteries.

myLens, 2016-01-11:
@AndreasAntener thank you for your clarification. In that case smart batteries are TOO smart for me and even the available smart-battery controllers (from TI at least) are not an appropriate onboard companion for a UAV imo. their main field seems to be charging controllers dedicated for laptops and similar devices, sensing current, .... this makes flying more risky than safer i guess :-( 
SM-Bus being the preferred means of communication shows tells the story by itself.
THANK YOU ANYWAYS !

closed 2016-05-22 by LorenzMeier