PX4/PX4-Autopilot #3366: two error when I make px4fmu-v2_default, my gcc version is 4.8, I don't know how to handle it.

R-bob, 2015-12-16:
make -C /home/xrb/src/Firmware/build_px4fmu-v2_default -j4 --no-print-directory 
[  0%] Built target __nuttx_patch_px4io-v2
[  0%] Built target mixer_gen
[  1%] Built target git_genmsg
[  1%] Built target git_gencpp
[  1%] Built target git_nuttx
[  1%] Built target __nuttx_patch_px4fmu-v2
[  1%] Built target git_mavlink
[  2%] Built target git_matrix
[  2%] Built target __nuttx_copy_px4io-v2
[  3%] Built target xml_gen
[  4%] Built target git_uavcan
Running from the source directory
[  4%] Built target msg_gen
[  4%] Built target __nuttx_copy_px4fmu-v2
[  5%] Built target nuttx_export_px4io-v2
[  6%] Built target nuttx_export_px4fmu-v2
[  6%] Built target io_prebuild_targets
[  6%] Built target prebuild_targets
[  6%] Built target platforms__common
[  7%] Built target drivers__stm32
[ 14%] Built target px4io-v2
[ 16%] Built target drivers__device
[ 16%] Built target drivers__stm32__adc
[ 16%] Built target drivers__px4fmu
[ 16%] Built target drivers__stm32__tone_alarm
[ 16%] Built target drivers__led
[ 18%] Built target drivers__px4io
[ 18%] Built target drivers__rgbled
[ 20%] [ 20%] Built target drivers__boards__px4fmu-v2
Built target drivers__mpu6000
[ 21%] Built target drivers__ms5611
[ 21%] [ 21%] Built target drivers__lsm303d
Built target drivers__mpu9250
[ 22%] Built target drivers__hmc5883
[ 22%] Built target drivers__l3gd20
[ 22%] Building CXX object src/drivers/srf02/CMakeFiles/drivers__srf02.dir/srf02.cpp.obj
[ 22%] Building CXX object src/drivers/mb12xx/CMakeFiles/drivers__mb12xx.dir/mb12xx.cpp.obj
[ 23%] Built target drivers__gps
/home/xrb/src/Firmware/src/drivers/srf02/srf02.cpp:56:18: fatal error: vector: No such file or directory
 #include <vector>
                  ^
compilation terminated.
src/drivers/srf02/CMakeFiles/drivers__srf02.dir/build.make:54: recipe for target 'src/drivers/srf02/CMakeFiles/drivers__srf02.dir/srf02.cpp.obj' failed
make[3]: *** [src/drivers/srf02/CMakeFiles/drivers__srf02.dir/srf02.cpp.obj] Error 1
CMakeFiles/Makefile2:1496: recipe for target 'src/drivers/srf02/CMakeFiles/drivers__srf02.dir/all' failed
make[2]: *** [src/drivers/srf02/CMakeFiles/drivers__srf02.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
/home/xrb/src/Firmware/src/drivers/mb12xx/mb12xx.cpp:58:18: fatal error: vector: No such file or directory
 #include <vector>
                  ^
compilation terminated.
src/drivers/mb12xx/CMakeFiles/drivers__mb12xx.dir/build.make:54: recipe for target 'src/drivers/mb12xx/CMakeFiles/drivers__mb12xx.dir/mb12xx.cpp.obj' failed
make[3]: *** [src/drivers/mb12xx/CMakeFiles/drivers__mb12xx.dir/mb12xx.cpp.obj] Error 1
CMakeFiles/Makefile2:1450: recipe for target 'src/drivers/mb12xx/CMakeFiles/drivers__mb12xx.dir/all' failed
make[2]: *** [src/drivers/mb12xx/CMakeFiles/drivers__mb12xx.dir/all] Error 2
[ 25%] Built target drivers__ll40ls
[ 25%] Built target libuavcan_dsdlc
Makefile:146: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:131: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 2

LorenzMeier, 2015-12-18:
Uninstall your current version:

```sudo apt-get remove gcc-arm-none-eabi```

then install the correct version:

```
pushd .
cd ~
wget https://launchpadlibrarian.net/186124160/gcc-arm-none-eabi-4_8-2014q3-20140805-linux.tar.bz2
tar -jxf gcc-arm-none-eabi-4_8-2014q3-20140805-linux.tar.bz2
exportline="export PATH=$HOME/gcc-arm-none-eabi-4_8-2014q3/bin:\$PATH"
if grep -Fxq "$exportline" ~/.profile; then echo nothing to do ; else echo $exportline >> ~/.profile; fi
. ~/.profile
popd
```

closed 2015-12-18 by LorenzMeier

badfirefly, 2016-04-13:
After install gcc-arm-eabi 4.8, got new errors:
/bin/sh: 1: /usr/bin/arm-none-eabi-g++: not found
ubuntu 14.04

julianoes, 2016-04-15:
Make sure your `.profile` is sourced. You can try to source it manually with:
```
source ~/.profile
```

It should also be sourced after you logout and log back in.