PX4/PX4-Autopilot #3368: SITL with gazebo and px4

alejodosr, 2015-12-17:
Hi,

I am trying to run this SITL simulation: https://pixhawk.org/dev/ros/sitl

However, Mavros is not receiving any HEARTBEAT (no connection despite UDP IP and ports seems to be correctly mapped). Do you have any idea?

Does anyone has successfully run a SITL simulation using Gazebo and px4?

LorenzMeier, 2015-12-17:
Use the guide on http://dev.px4.io instead

alejodosr, 2015-12-17:
You mean this one: http://dev.px4.io/simulation-gazebo.html ?
Because it seems to be incomplete, this make does not found the target: make posix_sitl_default gazebo

LorenzMeier, 2015-12-17:
It does work. Update your Firmware repository. I have used this today.

closed 2015-12-17 by LorenzMeier

LorenzMeier, 2016-01-02:
http://dev.px4.io/simulation-ros-interface.html

tuofeichen, 2016-01-25:
Sorry, I run into the same problem. Lemme know if I missed anything,
0. Have all dependencies installed (gazebo, ros and everything), and have roslaunch mavros px4.launch tested out with hardware (a pixhawk) and received heart beat. 
1. make posix_sitl_default gazebo: compiled and run, showed simulation window just fine
2. roslaunch mavros px4.launch fcu_url:="udp://:14540@192.168.1.36:14557"

The mavros node doesn't get a heartbeat and the simulator constantly output a mavlink sim timeout for 10 ms. 

I've tried: 
1, change the url to the local host one, doesn't work
2. change the port number to 14556, doesn't work. 

Thank you so much

Best
Francis

devbharat, 2016-01-26:
You don't need ROS or mavros for the native gazebo sitl that you start with  make posix_sitl_default gazebo. No communication here happens over ROS here and so mavros sees no data. You should not see the "mavlink sim timeout for 10 ms" through, if both gazebo and px4 are running fine.

For running the px4 sitl with ROS, start the px4 sitl with no_sim=1, 

    no_sim=1 make posix_sitl_default gazebo

then it waits for sensor data from 'some' simulator on udp port. Then, if you have PX4/rotor_simulator built with mavlink_interface, you can launch it for example with such a launch file:

    <launch>

        <arg name="headless" default="false"/>
        <arg name="gui" default="true"/>
        <arg name="enable_logging" default="false"/>
        <arg name="enable_ground_truth" default="true"/>
        <arg name="enable_mavlink_interface" default="true"/>
        <arg name="ns" default="iris"/>
        <arg name="log_file" default="$(arg ns)"/>
        <arg name="mavlink_bridge_url" default="udp://localhost:14553@localhost:14560" />

        <include file="$(find rotors_gazebo)/launch/iris_empty_world.launch">
            <arg name="headless" value="$(arg headless)"/>
            <arg name="gui" value="$(arg gui)"/>
            <arg name="enable_logging" value="$(arg enable_logging)" />
            <arg name="enable_ground_truth" value="$(arg enable_ground_truth)" />
            <arg name="enable_mavlink_interface" value="$(arg enable_mavlink_interface)" />
            <arg name="log_file" value="$(arg log_file)"/>
        </include>
        <include file="$(find mavros)/launch/mavlink_bridge.launch">
        <arg name="mavlink_bridge_url" value="$(arg mavlink_bridge_url)" />
        <arg name="ns" value="$(arg ns)"/>
        </include>

    </launch>

This uses mavros to bridge ROS data to UDP for px4 sitl. now you can do all ROS stuff.

tuofeichen, 2016-01-27:
Hi 
Thank you so much for the response. A couple si more things if you can help out
1. I still can't figure out the mavlink time out error, this is literally what I did, did you see anything wrong? Or should I redownload the Firmware from github?

```
Script started on Tue 26 Jan 2016 06:41:53 PM CST
]0;parallels@ubuntu: make posix_sitl_default gazebo
git submodule init
Tools/check_submodules.sh 
Git submodule config valid.
Checked NuttX submodule, correct version found
Checked Tools/gencpp submodule, correct version found
Checked Tools/genmsg submodule, correct version found
Checked Tools/jMAVSim submodule, correct version found
Checked Tools/sitl_gazebo submodule, correct version found
Checked cmake/cmake_hexagon submodule, correct version found
Checked mavlink/include/mavlink/v1.0 submodule, correct version found
Checked src/lib/DriverFramework submodule, correct version found
Checked src/lib/dspal submodule, correct version found
Checked src/lib/ecl submodule, correct version found
Checked src/lib/matrix submodule, correct version found
Checked src/modules/uavcan/libuavcan submodule, correct version found
Checked unittests/googletest submodule, correct version found
ninja -C /home/parallels/src/Firmware/build_posix_sitl_default  gazebo
ninja: Entering directory `/home/parallels/src/Firmware/build_posix_sitl_default'

[1/10] cd /home/parallels/src/Firmware && git submodule update --recursive Tools/genmsg[K
[2/10] cd /home/parallels/src/Firmware && git submodule update --recursive Tools/gencpp[K
[3/10] cd /home/parallels/src/Firmware && git ...pdate --recursive mavlink/include/mavlink/v1.0[K
[3/10] cd /home/parallels/src/Firmware && git submodule update --recursive Tools/gencpp[K
[4/10] cd /home/parallels/src/Firmware && git submodule update --recursive src/lib/matrix[K
[4/10] cd /home/parallels/src/Firmware && git submodule update --recursive Tools/genmsg[K
[5/10] cd /home/parallels/src/Firmware && git submodule update --recursive Tools/sitl_gazebo[K
[5/10] cd /home/parallels/src/Firmware && git ...pdate --recursive mavlink/include/mavlink/v1.0[K
[6/10] cd /home/parallels/src/Firmware && git submodule update --recursive src/lib/ecl[K
[6/10] cd /home/parallels/src/Firmware && git submodule update --recursive src/lib/matrix[K
[7/10] cd /home/parallels/src/Firmware && git submodule update --recursive Tools/jMAVSim[K
[7/10] cd /home/parallels/src/Firmware && git submodule update --recursive Tools/sitl_gazebo[K
[8/10] cd /home/parallels/src/Firmware && git ...ule update --recursive src/lib/DriverFramework[K
[8/10] cd /home/parallels/src/Firmware && git submodule update --recursive src/lib/ecl[K
[9/10] cd /home/parallels/src/Firmware && git ...pdate --recursive src/modules/uavcan/libuavcan[K
[9/10] cd /home/parallels/src/Firmware && git submodule update --recursive Tools/jMAVSim[K
[9/10] cd /home/parallels/src/Firmware && git ...ule update --recursive src/lib/DriverFramework[K
[9/10] cd /home/parallels/src/Firmware && git ...pdate --recursive src/modules/uavcan/libuavcan[K
[10/10] cd /home/parallels/src/Firmware && Too...arallels/src/Firmware/build_posix_sitl_default[K
SITL ARGS
rc_script: posix-configs/SITL/init/rcS
debugger: none
program: gazebo
model: none
build_path: /home/parallels/src/Firmware/build_posix_sitl_default
empty model, setting iris as default
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   thread
-- Configuring done
-- Generating done
-- Build files have been written to: /home/parallels/src/Firmware/Tools/sitl_gazebo/Build
make[1]: Entering directory `/home/parallels/src/Firmware/Tools/sitl_gazebo/Build'
make[2]: Entering directory `/home/parallels/src/Firmware/Tools/sitl_gazebo/Build'
make[3]: Entering directory `/home/parallels/src/Firmware/Tools/sitl_gazebo/Build'
make[3]: Entering directory `/home/parallels/src/Firmware/Tools/sitl_gazebo/Build'
make[3]: Leaving directory `/home/parallels/src/Firmware/Tools/sitl_gazebo/Build'
make[3]: Leaving directory `/home/parallels/src/Firmware/Tools/sitl_gazebo/Build'
[ 67%] Built target mav_msgs
make[3]: Entering directory `/home/parallels/src/Firmware/Tools/sitl_gazebo/Build'
make[3]: Entering directory `/home/parallels/src/Firmware/Tools/sitl_gazebo/Build'
[ 72%] Built target hello_world
make[3]: Leaving directory `/home/parallels/src/Firmware/Tools/sitl_gazebo/Build'
[... 368 lines cut ...]
started core service [/rosout]
process[gazebo-2]: started with pid [22350]
process[gazebo_gui-3]: started with pid [22353]
process[spawn_iris-4]: started with pid [22359]
process[iris/mavlink_bridge-5]: started with pid [22380]
[ INFO] [1453856314.510807377]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1453856314.513966270]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
spawn_model script started
[INFO] [WallTime: 1453856315.057270] [0.000000] Loading model xml from ros parameter
[INFO] [WallTime: 1453856315.064092] [0.000000] Waiting for service /gazebo/spawn_urdf_model
[INFO] [WallTime: 1453856315.972884] [0.000000] Calling service /gazebo/spawn_urdf_model
[ INFO] [1453856316.076229950, 0.025000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[INFO] [WallTime: 1453856316.441446] [0.250000] Spawn status: SpawnModel: Successfully spawned model
[ INFO] [1453856316.507207347, 0.250000000]: Physics dynamic reconfigure ready.
[spawn_iris-4] process has finished cleanly
log file: /home/parallels/.ros/log/163b96a8-c491-11e5-9f11-001c429eb260/spawn_iris-4*.log
^C[iris/mavlink_bridge-5] killing on exit
```

The gazebo opens up but the SITL doesn't seem to react to it (still waiting for simulator to open). Any idea why is this happens?

Thank you so much.

Best
Francis

julianoes, 2016-01-27:
@tuofeichen Try this:

```
 roslaunch mavros px4.launch fcu_url:="udp://:14550@127.0.0.1:14557"
```

tuofeichen, 2016-01-27:
Hey @julianoes 
Thanks, that does the trick.

tuofeichen, 2016-01-28:
Hey guys,
Sorry for the follow-up questions, complete noob here. I still can't quite figure out why the mavlink sim timeout warning continues to happen. I have the master node by launching the px4.launch, and initially I think it might be just I'm not sending velocity command to it, but even when I did, using a simple publisher, the drone is able to take off in gazebo but the simulator still keeps spitting out the mavlink sim timeout warning. 

Any idea? Or do you guys want to see any part of my code?

Thanks

Best
Tuofei

mhkabir, 2016-01-28:
Your computer is too slow. You can mostly ignore the warnings.

rs1990, 2016-08-03:
Hello,

Iv been trying to get the simulator working but everytime i run ```` make posix gazebo_iris_opt_flow ````  i end up with the following error,

````
swapnil@UTARIROS08:~/catkin_ws/src/px4_firmware/Firmware$ make posix gazebo_iris_opt_flow
PX4 CONFIG: posix_sitl_default
[  1%] [  1%] Built target git_gencpp
Built target prebuild_targets
[  1%] [  2%] Built target git_genmsg
Built target git_uavcan
[  2%] Built target git_mavlink
[  3%] [  3%] Built target git_jmavsim
Built target git_matrix
[  5%] Built target df_driver_framework
[  5%] Built target git_driverframework
[  5%] Built target git_ecl
[  5%] [  6%] Built target git_gazebo
Built target xml_gen
[  7%] Built target mixer_gen
[ 31%] Built target msg_gen
[ 31%] [ 31%] Built target generate_topic_listener
Built target platforms__common
[ 32%] Built target drivers__boards__sitl
[ 33%] Built target drivers__gps
[ 35%] Built target drivers__device
[ 35%] Built target drivers__pwm_out_sim
[ 36%] Built target platforms__posix__drivers__adcsim
[ 44%] Built target systemcmds__tests
[ 44%] Built target platforms__posix__drivers__accelsim
[ 44%] Built target platforms__posix__drivers__airspeedsim
[ 44%] Built target platforms__posix__drivers__barosim
[ 44%] Built target platforms__posix__drivers__gyrosim
[ 45%] Built target platforms__posix__drivers__gpssim
[ 45%] Built target platforms__posix__drivers__ledsim
[ 46%] Built target platforms__posix__drivers__tonealrmsim
[ 46%] Built target platforms__posix__drivers__rgbledsim
[ 46%] Built target systemcmds__esc_calib
[ 47%] Built target platforms__posix__px4_layer
[ 47%] [ 50%] [ 52%] Built target systemcmds__param
Built target platforms__posix__work_queue
Built target systemcmds__mixer
[ 53%] [ 53%] Built target systemcmds__reboot
Built target systemcmds__sd_bench
[ 53%] [ 54%] Built target systemcmds__perf
Built target systemcmds__topic_listener
[ 54%] [ 54%] [ 54%] Built target systemcmds__ver
Built target systemcmds__top
Built target systemcmds__motor_ramp
[ 54%] Built target modules__attitude_estimator_q
[ 54%] [ 54%] Built target modules__attitude_estimator_ekf
Built target modules__dataman
[ 54%] Built target modules__ekf2
[ 58%] Built target modules__commander
[ 59%] [ 59%] Built target lib__external_lgpl
Built target modules__ekf_att_pos_estimator
[ 60%] Built target modules__land_detector
[ 61%] Built target modules__mc_att_control_multiplatform
[ 66%] [ 68%] Built target lib__ecl
Built target modules__uORB
[ 68%] Built target modules__mc_pos_control
[ 69%] Built target modules__mc_att_control
[ 69%] Built target modules__param
[ 70%] Built target modules__mc_pos_control_multiplatform
[ 71%] Built target modules__position_estimator_inav
[ 77%] [ 77%] Built target modules__local_position_estimator
Built target modules__mavlink
[ 81%] Built target modules__navigator
[ 82%] Built target modules__sdlog2
[ 82%] Built target modules__replay
[ 82%] Built target modules__sensors
[ 83%] Built target modules__simulator
[ 84%] Built target modules__systemlib__mixer
[ 85%] Built target modules__vtol_att_control
[ 90%] Built target modules__systemlib
[ 90%] Built target lib__conversion
[ 91%] Built target lib__controllib
[ 91%] [ 91%] Built target lib__geo_lookup
Built target lib__DriverFramework__framework
[ 92%] [ 92%] Built target lib__geo
Built target lib__mathlib__math__filter
[ 93%] Built target lib__mathlib
[ 93%] Built target lib__launchdetection
[ 94%] Built target lib__rc
[ 94%] Built target lib__tailsitter_recovery
[ 94%] Built target lib__runway_takeoff
[ 94%] Built target lib__terrain_estimation
[ 94%] Built target examples__px4_simple_app
[ 95%] Built target drivers__sf0x__sf0x_tests
[ 96%] Built target lib__rc__rc_tests
[ 97%] Built target modules__commander__commander_tests
[ 97%] [ 97%] Built target modules__controllib_test
Built target modules__unit_test
[ 98%] Built target modules__uORB__uORB_tests
[100%] [100%] [100%] Built target modules__logger
Built target modules__fw_pos_control_l1
Built target modules__fw_att_control
[100%] Built target px4
SITL ARGS
rc_script: posix-configs/SITL/init/rcS
debugger: none
program: gazebo
model: iris_opt_flow
build_path: /home/swapnil/catkin_ws/src/px4_firmware/Firmware/build_posix_sitl_default
make[5]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
make[8]: *** No rule to make target `/home/swapnil/catkin_ws/src/px4_firmware/Firmware/Tools/sitl_gazebo/PROTOBUF_PROTOC_EXECUTABLE-NOTFOUND', needed by `MotorSpeed.pb.cc'.  Stop.
make[7]: *** [CMakeFiles/mav_msgs.dir/all] Error 2
make[7]: *** Waiting for unfinished jobs....
[  4%] Built target sdf
make[6]: *** [all] Error 2
make[5]: *** [gazebo_build] Error 2
make[4]: *** [src/firmware/posix/CMakeFiles/gazebo_iris_opt_flow] Error 2
make[3]: *** [src/firmware/posix/CMakeFiles/gazebo_iris_opt_flow.dir/all] Error 2
make[2]: *** [src/firmware/posix/CMakeFiles/gazebo_iris_opt_flow.dir/rule] Error 2
make[1]: *** [gazebo_iris_opt_flow] Error 2
make: *** [posix_sitl_default] Error 2
swapnil@UTARIROS08:~/catkin_ws/src/px4_firmware/Firmware$ 
```` 

could someone help me please ? Thanks a lot. Im using ubuntu 14.04 and i have gazebo 7 installed.

YuYangguang, 2017-10-13:
I also cannot receive any data from mavros after ran no_sim=1 make posix_sitl_default gazebo , did you solve this issue? Thank you so much.