PX4/PX4-Autopilot #3370: Question regarding dynamics model in BlockLocalPositionEstimator

ecmnet, 2015-12-18:
Might be that I do not understand all the math or the code behind BlockLocalPositionEstimator. Maybe one of you guys can help me out here with a short answer. As far as I understood, BlockLocalPositionEstimator uses accelerator data to update the covariance matrix by speeds, not integrating to position. Is that assumption correct?

mhkabir, 2015-12-18:
@jgoppert

jgoppert, 2015-12-18:
@ecmnet The acceleration data is used to predict the evolution of velocity and position, via integration. The position and velocity are then corrected by other sensors.

ecmnet, 2015-12-18:
@jgoppert Thank you for the info. I clearly understand the integration of accelerator data to velocity in BlockLocalPositionEstimator; but I am curious how the integration to position should work...

LorenzMeier, 2016-05-22:
Outdated.

closed 2016-05-22 by LorenzMeier