PX4/PX4-Autopilot #3372: Allow small VTOL operation without airspeed sensor

LorenzMeier, 2015-12-18:
This is highly wanted and probably needed for the Caipiroshka.

sanderux, 2016-01-12:
@tumbili 
Would this not be inherent when VT_ARSP_TRANS is set to 0? 
https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/tailsitter.cpp#L180

melisab, 2016-02-14:
I was wondering if replacing airspeed sensor means that we can rely on mc altitude control to see if the plane has achieved a sufficient lift to safely declare that it's in FW mode.

SimonWilks, 2016-02-14:
Now that's an interesting idea :) I wonder if this could at least be used as a sanity check for airspeed failure detection ( #3746  ).

sanderux, 2016-02-14:
@tumbili @AndreasAntener could we safely increase the altitude setpoint for FW by something significant and check if transition has occurred by reaching this altitude? as @SimonWilks stated, this could serve for both the lack of, and a malfunctioning airspeed sensor.

tumbili, 2016-02-19:
@LorenzMeier Is this about allowing at least to hover or about allowing both regimes?
We could just use gps speed for the transitions or open loop based on time.

sanderux, 2016-02-19:
@simonwilks did you mean to close this issue?

SimonWilks, 2016-02-19:
No. Sorry ;)

LorenzMeier, 2016-02-19:
Full operation of course. A VTOL in hover only is just a very bad quad.

tumbili, 2016-06-08:
Implemented in #4770

closed 2016-06-08 by tumbili