PX4/PX4-Autopilot #3374: Intermittent failure to generate PWM signals after arming

kd0aij, 2015-12-19:
Observed with both AUAV-X2 in a VTOL config, and on PixRacer quadcopter. Sometimes motors don't spin after arming. Verified no PWM output from PixRacer when this occurs.

LorenzMeier, 2015-12-19:
Do you have safety disabled on the X2?

kd0aij, 2015-12-19:
No the X2 has a safety switch. Can't reproduce the problem on my PixRacer with either beta or master. @pkcomoud says his motors never start on beta, intermittent on master.

LorenzMeier, 2015-12-20:
Since master == beta right now that report is confusing. Probably some init related issue in the FMU driver.

kd0aij, 2015-12-20:
Re: https://gitter.im/PX4/Firmware?at=5676be13f31bbe91555b647d
My testing shows the same PWM output on bootup with radio on and off. Signal goes high for about 300msec before pulsetrain begins. Perhaps it should initialize to low?

pkocmoud, 2016-01-27:
I think we had a user with this issue. I was told that was resolved by a code change by Lorenz. Should this be closed?

kd0aij, 2016-01-27:
I haven't flown the X2 VTOL in a while, and since you're not seeing it anymore will close this for now.

closed 2016-01-27 by kd0aij