PX4/PX4-Autopilot #3375: FMUv4 motors begin spinning at start of arming tone instead of at end of tone.

kd0aij, 2015-12-19:


LorenzMeier, 2016-01-01:
I've assigned this to yourself. There are two potential root causes:

  * A too high DISARMED PWM or MIN_PWM setting
  * An error in the ramp logic

The offending code is somewhere here:
https://github.com/PX4/Firmware/blob/master/src/drivers/px4fmu/fmu.cpp#L771-L773

Most likely the ramp doesn't get triggered correctly. You can use printfs / warnx there since the driver is not running in interrupt context.

kd0aij, 2016-01-02:
additional observation: motors start at end of tone on first arming event after reboot, but at beginning of tone on subsequent arming events

LorenzMeier, 2016-01-02:
So you just need to check why the state doesn't get reset on disarm.

kd0aij, 2016-01-03:
[pwm_limit_state_machine.pdf](https://github.com/PX4/Firmware/files/76633/pwm_limit_state_machine.pdf)

LorenzMeier, 2016-01-03:
Nice! Could you add that to the dev guide? It supports diagrams, in this format:

```
```mermaid
graph LR;
  Gazebo-->Plugin;
  Plugin-->MAVLink;
  MAVLink-->SITL;
```
```

http://knsv.github.io/mermaid/#usage

LorenzMeier, 2016-01-03:
Example here: https://raw.githubusercontent.com/PX4/Devguide/master/book/simulation-gazebo.md

kd0aij, 2016-01-04:
fmu was asserting both armed and pre_armed inputs to pwm_limit_calc even when !_armed.armed
mermaid doesn't seem to render properly in Chromium or Firefox

LorenzMeier, 2016-01-04:
Have you fixed that in your PR accordingly? All works now?

kd0aij, 2016-01-04:
yes, just commented the PR...

closed 2016-01-06 by LorenzMeier