PX4/PX4-Autopilot #3381: Simulating px4flow in SITL

hridaybavle, 2015-12-21:
Hi, 
I followed the following guide for Pixhawk SITL
http://dev.px4.io/simulation-gazebo.html. 
I am able to read and send data using mavros. But now I want the altitude and the speeds data from the optical flow sensor. 

When I launch "make posix_sitl_default gazebo", I see the message "stream OPTICAL_FLOW_RAD on UDP port 14556 enabled with rate 10.0 Hz", which I guess means that the optical flow data is being published. But when I check the topic mavros/px4flow/raw/optical_flow_rad there is not data being published. 

Do I need to do something in order to start receiving the simulated optical flow data?

LorenzMeier, 2016-03-24:
This works on master if you use the flow simulation startup as documented on http://dev.px4.io

closed 2016-03-24 by LorenzMeier

hridaybavle, 2016-03-29:
Hi, 
I followed this tutorial for setting up SILT
http://dev.px4.io/simulation-gazebo.html
and I use the following command in order to receive the optical flow data
**make posix gazebo_iris_opt_flow**

But I do not receive any data and get the following output

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 6.5.1
Copyright (C) 2012-2015 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.0.67
[Wrn] [gazebo_lidar_plugin.cpp:78] [gazebo_lidar_plugin] Please specify a robotNamespace.
[Wrn] [gazebo_opticalFlow_plugin.cpp:85] [gazebo_optical_flow_plugin] Please specify a robotNamespace.
Publishing to: motors
INFO  Sending initial controls message to simulator

dataman: Unknown restart, data manager file 'rootfs/fs/microsd/dataman' size is 103090 bytes

927102 Added driver 0xced120 /dev/rgbled0
927117 Added driver 0xced3e0 /dev/tone_alarm0
INFO  TONE_SET_ALARM 0

INFO  GYROSIM::init
927147 Added driver 0xced6d0 /dev/accel0
927154 Added driver 0xcedad0 /dev/gyro0
INFO  GYROSIM_gyro::init base class ret: 0

927209 Added driver 0xcee5f0 /dev/accel1
927217 Added driver 0xcee880 /dev/mag0
927261 Added driver 0xcef470 /dev/baro0
947351 Added driver 0xcefb50 /dev/adc0
947371 Added driver 0xcefe00 /dev/gpssim

pwm_out_sim MODE_8PWM

[Wrn] [RaySensor.cc:312] ranges not constructed yet (zero sized)
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.0.67
[Err] [InsertModelWidget.cc:298] Missing model.config for model "/home/cvg/workspace/Firmware/Tools/sitl_gazebo/models/rotors_description"
[Wrn] [ModelDatabase.cc:211] Unable to connect to model database using [//database.config]. Only locally installed models will be available.
WARN  pwm_out_sim: No valid fds

INFO  GYROSIM::_set_sample_rate 1000003 Hz
INFO  GYROSIM: Changed sample rate to 500Hz
INFO  GYROSIM::_set_sample_rate 1000003 Hz
INFO  GYROSIM: Changed sample rate to 500Hz

INFO  TONE_SET_ALARM 1

ERROR px4_task_spawn_cmd: failed to set name of thread 34 0

ERROR px4_task_spawn_cmd: failed to set name of thread 34 0

WARN  mode: 0, data rate: 2000000 B/s on udp port 14556
WARN  offboard mission init: ERROR
INFO  Setting up UDP w/port 14556

WARN  set remote port 14540
WARN  mode: 2, data rate: 2000000 B/s on udp port 14557
INFO  Setting up UDP w/port 14557

WARN  stream POSITION_TARGET_LOCAL_NED on UDP port 14556 enabled with rate 80.0 Hz

WARN  stream LOCAL_POSITION_NED on UDP port 14556 enabled with rate 80.0 Hz

WARN  stream GLOBAL_POSITION_INT on UDP port 14556 enabled with rate 80.0 Hz

WARN  stream ATTITUDE on UDP port 14556 enabled with rate 80.0 Hz

WARN  stream ATTITUDE_TARGET on UDP port 14556 enabled with rate 80.0 Hz

WARN  stream RC_CHANNELS on UDP port 14556 enabled with rate 20.0 Hz

WARN  stream HIGHRES_IMU on UDP port 14556 enabled with rate 250.0 Hz

WARN  stream OPTICAL_FLOW_RAD on UDP port 14556 enabled with rate 10.0 Hz

WARN  log buffer size: 8192 bytes
INFO  [blackbox] rootfs/fs/microsd/log/2016-03-29
WARN  sdlog2: failed setting sched params
INFO  [blackbox] recording: 16_41_05.px4log
WARN  changing partner IP to: 127.0.0.1
WARN  pwm_out_sim: No valid fds
WARN  INAV poll timeout
WARN  Q POLL TIMEOUT
Command 'sdlog2' failed, returned 1

I am able to receive the data properly when using 
**make posix_sitl_default gazebo**

 #3714 has the same error, but there does not seem to be a proper solution. I am using the firmware from the latest master and I tested on two computers but with the same results. 

Any help would be appreciated. 

Thanks.

vinhk, 2016-07-20:
@hridaybavle did you get the topics for the optical flow to work?

hridaybavle, 2016-07-20:
We implemented the px4flow from the camera image and gyro data from the IMU. I can check we can share the code.

vinhk, 2016-07-20:
okay, that sounds good. I have been wondering why I can not get the px4flow topic to display information on the terminal.