PX4/PX4-Autopilot #3385: Activate takeoff/land when in alt/posctl on extreme throttle positions

AndreasAntener, 2015-12-24:


AndreasAntener, 2016-01-04:
@kd0aij if implemented, you don't need to use the kill switch anymore for landings ;) (in landing mode it will disarm very quickly on the ground)

AndreasAntener, 2016-02-16:
Thinking about the landing: just on extreme positions probably doesn't work, maybe we can just extend the ground detection to trigger also when not in landing.

julianoes, 2016-02-16:
So for takeoff: what about blocking takeoff until a user goes all the way to the bottom with throttle and then goes back up. It would be similar to the way you have to initialize any kind of cheap toy drone or FPV racer.

kd0aij, 2016-02-16:
@AndreasAntener That's good unless it disarms in the air accidentally. Doesn't eliminate the need for a kill switch though, because we don't know when the pilot might want to stop the motors due to an emergency.

kd0aij, 2016-02-16:
Thinking about that a little more, is it guaranteed that a pilot wants to disarm after setting down? Perhaps not?

LorenzMeier, 2016-02-27:
@AndreasAntener Could you take a stab at this, given you likely understand the issue at hand best?

jgoppert, 2016-02-27:
I think that throttle not responding in alt/pos-ctl mode is very unnerving and if this is a new feature I think it should be removed.

Is this all just so that the vehicle spins down when doing auto landings? If that is the case we should make the spin down conditional on the throttle stick not being high.

mhkabir, 2016-02-27:
I agree with James. It's really weird to see this behaviour. When did this come in? I remember being used to ALTCTL takeoffs.

LorenzMeier, 2016-02-27:
You're wrong. We just removed the requirement to arm in manual. Nothing broke. It just needs to be adjusted to the new arming regime.

mhkabir, 2016-02-27:
Arming in manual and switching to ALTCTL does not give me throttle control.

AndreasAntener, 2016-02-29:
Let me explain where this comes from:
**a)** spring loaded throttle that is at 50%, by default flying in position or altitude control and having a button for takeoff/land
**b)** altitude and position control with GPS and baro on the ground are rather problematic because of position shifts, it's quite easy to tip over your quad if you just leave it sitting for a while

To resolve the issues James and Kabir are facing it would be relatively simple to just exit the current idle condition when throttle get's higher. But this doesn't work for a spring loaded throttle stick. So the idea here was that we would enter auto takeoff to 2.5m when you go full throttle. But I'm guessing that this won't be satisfying for guys either because you expect manual control.

BTW: @jgoppert the spin down on landing is independent

I don't see a solution to **a)** that would work for spring loaded and non-spring loaded throttle sticks at the same time (ideas? :)). Maybe we need a parameter to specify that. And the feature this issue describes would only apply for spring loaded throttle. Then we can add the condition to exit idle state as soon as you go over X with throttle.

Thoughts?

LorenzMeier, 2016-02-29:
I don't see the issue here. We should just take off if the throttle is maxed (using the takeoff mode). If the user continues to push up the throttle it will switch to POSCTRL and continue to rise.

Same thing for down throttle: If the throttle is all the way down and landing detected, disarm after a few seconds.

I don't see the corner cases for any throttle position.

mhkabir, 2016-02-29:
I agree with Lorenz. Don't quite see the issue here. 

> altitude and position control with GPS and baro on the ground are rather problematic because of position shifts, it's quite easy to tip over your quad if you just leave it sitting for a while

Regarding that, we can just stop controlling position or altitude if we on the ground (still remaining in the mode, but not trying to control alt/pos).  I think land detection is robust enough for that. Just keep reseting the setpoint while still on ground. 

Then, when the stick is brought to max (while on ground) takeoff mode is activated. There should not be any response to throttle other than max while on the ground (like what it's now ; what James and I faced) 

No corner cases as far as I can see :) @jgoppert what do you think? This should be good for our indoor flying.

AndreasAntener, 2016-02-29:
> I don't see the issue here. We should just take off if the throttle is maxed (using the takeoff mode). If the user continues to push up the throttle it will switch to POSCTRL and continue to rise.

But what would be the exit condition for the takeoff mode? At the moment this is AGL (MIS_TAKEOFF_ALT) which will make a hole in Kabir and James ceiling if set wrong ;). We can't just use the takeoff state because it will already trigger when thrust > 0.15

jgoppert, 2016-02-29:
@AndreasAntener I'm glad to hear you guys are working on the 50% throttle spring type of controller. I just got a request for that from a customer. I think we want to make the behavior as close to DJI as possible as this is what people are used to now. This would basically mean if the user takes off in position control, then we start the AGL at 0 and increase as they push up the throttle. There is no reason we should go into mission mode where the MIS_TAKEOFF_ALT comes into play.

AndreasAntener, 2016-02-29:
@jgoppert "auto takeoff" isn't mission but a flight mode. It already works the way you wantit: 
- your in altitude or position control
- takeoff command (button or full throttle)
- switch to takeoff mode and climb to X m
- automatically switch back to the mode you were in before

The only question now is when to switch back to the mode you were in.

AndreasAntener, 2016-02-29:
Hm, on a second though, the trigger to exit takeoff could be when we reach the takeoff velocity. And we could probably implement this takeoff state relatively easy in the position controller without the takeoff flight mode.

tubeme, 2016-04-17:
I agree with @jgoppert DJI established pretty much the standard in take off and landing behavior that the users expect to see. 

Naza 2 procedure take off in GPS mode, Atti mode = Position Control, Altitude Control:

Arm
Props spin idle
Start to push the throttle until the middle nothing happen (this is applicable for a throttle with a spring). The moment you reach 50+% throttle the props start to spin faster and give the MC little push (like auto take off) until it reaches 2-3 meters then it follows your throttle stick.

Actually Naza 2 take off is unique because you can take off from uneven surface. I tried taking off from 15% slope and it takes off perfect, After I've reached 51% throttle it first balances the quad and then gives it a little punch until 2 meters and then gives me the control.

The same procedure could be implemented for the airplanes as well. When we arm in ALt Pos Control there is idle spin. The moment we reach 51% throttle it gives the plane auto throttle punch until the plane takes off, then it gives you back the throttle. The only thing should be considered, how we trigger abort of take off if we want to stop the plane in emergency.

For the landing the DJI stops the motors almost immediately if it feels it touched th ground and the throttle is at 0%. This happens in both guided modes. One of the most annoying things with PX4 for me was landing in Position control, having my quad dance on the ground, because at 0% throttle in Pos Control actually is a rate of descend which gives the quad the most dangerous lift, given the vortexes under the props I've had a lot of flips and crashes long time ago until I started preparing myself to flip the mode switch to manual the moment I touch down and then issue disarm command with the stick. 

 It is very useful if we could land the airplane in Alt Pos Control as well. Supposedly land with throttle 0% which gives us minimum airspeed and then once touch down detected to stop the prop. 

Also flying in Alt Pos Control turning on the flaps, could trigger landing mode in such a way that the min airspeed becomes FW_LND_AIRSPD_SC. This way intuitively the plane will be in a semi auto mode during Alt Pos Control landing. Once it touches the ground it should stop the prop.

tops4u, 2016-05-23:
I would support @tubeme as I had a Flip on the Ground as well in PosCtl were also GNSS Signal went bad. PosCtl detected a position deviation while sitting on the Ground and wanted to correct this. In the end the Copter tipped over on the ground breaking some Props. Landing with always flipping the Switch to a Manual Mode is a bit annoying at least when coming from a NAZA (which of course has other deficiencies).

AndreasAntener, 2016-11-21:
@tumbili could we pull this out and push? Before @julianoes finishes the MC refactoring :D

tubeme, 2016-12-03:
We've had a very bad situation today. We wanted to try the arming in Position and once we armed in Position the VTOL just shoot out in the sky full throttle without giving any throttle with the stick and it did not stop until we commanded Manual. The throttle lever was in 0 position. Do you need the LOG? We have to be very careful with this because it can harm somebody. Also there was no way we could DISARM in Position and Altitude modes. This is crazy. You land, and you cannot disarm and if at this moment there is some minor GPS uncertainty, it sure makes a flip on the ground. It is absurd situation that you have to move the switch to Manual and then give a disarm command. The disarming was like this before, but we thought this was corrected since 1.5. But neither the arming nor the disarming are well thought in Position and Altitude so far. Far far away from the DJI smoothness in this regard.

dagar, 2016-12-03:
@tubeme yes please share the log in a new issue.

LorenzMeier, 2016-12-04:
This particular issue is likely solved on master already.

western-co-at, 2016-12-12:
@julianoes you did a great work!
I rebased your refactor_mc_pos_control branch with the current master and did some test flights. It's now quite similar to start and land in pos control mode as with DJI flight controllers. It didn't require any nick and roll inputs for starting and landing. Also starting/landing in windy conditions were no problem. When landing on a slightly sloped ground (e.g. when landing with a small copter on a stone) it took some more seconds after the copter reduced the throttle and then disarmed.

tops4u, 2016-12-12:
I tried this only two times and got the following issues:

- Starting probably got a little YAW punch due to motor starting moment. So 3 out of 6 Motors stopped spinning while still on the ground and one started spinning up. I stopped the start and started in Stabilized.

- Landing on a uneven surface (foot path) caused the Copter to try an compensate for this by spinning the lowest Motor up. I was unable to disarm - probably the landing was not detected, even with very low power to the remaining motors. So I used the Kill-Switch in order to prevent a On-the-Ground Flipover. 

So maybe this only works with perfect conditions.

julianoes, 2017-02-13:
@AndreasAntener I would say this is implemented now, right?

LorenzMeier, 2017-02-13:
Yep it is.

closed 2017-02-13 by LorenzMeier