PX4/PX4-Autopilot #3391: DSM input parsing on FMUv4

LorenzMeier, 2015-12-26:


cctsao1008, 2016-01-03:
From fmu.cpp, I see these

#ifdef DSM_SERIAL_PORT
		// XXX rather than opening it we need to cycle between protocols until one is locked in
		//_dsm_fd = dsm_init(DSM_SERIAL_PORT);
#endif

Does it means the DSM input disabled on FMUv4 , right?

LorenzMeier, 2016-01-03:
I wouldn't say disabled, its just not implemented but will be soon.

LorenzMeier, 2016-01-03:
@kd0aij I've assigned you here - if you are willing to take this on we should talk through the architecture of it first.

kd0aij, 2016-01-03:
OK. Should be a trivial matter to read data from a serial port.

LorenzMeier, 2016-01-03:
What you need to do is something like this:
```
enum RC_SCAN {
RC_SCAN_PPM = 0,
RC_SCAN_SBUS,
RC_SCAN_DSM,
RC_SCAN_SUMD,
RC_SCAN_ST24
}
```
then in the main loop

uint64_t rc_scan_last_lock = 0;
uint64_t rc_scan_begin = 0;
bool rc_scan_locked = false;
enum RC_SCAN _rc_scan_state = RC_SCAN_PPM;

// Scan for one second, then switch protocol
constexpr uint64_t rc_scan_max = 1000 * 1000;

```
switch (_rc_scan_state) {
  case RC_SCAN_SBUS:
  if (rc_scan_begin == 0) {
  // Configure S.BUS port
  stm32_gpio, etc.
  } else if (hrt_absolute_time() - rc_scan_last_lock < rc_scan_max || hrt_absolute_time() - rc_scan_begin < rc_scan_max) {
  // read port
  if (read_success) {
    rc_scan_last_lock = hrt_absolute_time();
 } else {
  rc_scan_begin = 0;
  _rc_scan_state++;
}
  
  case RC_SCAN_ST24:
   // If scan interval exceeded (else case)
  _rc_scan_state = RC_SCAN_PPM; // go back to first entry in scan list
}
```

The associated code is here:
https://github.com/PX4/Firmware/blob/master/src/modules/px4iofirmware/controls.c#L74-L150

What is a bit odd about that is that the DSM code reads all bytes and then all protocols parse it. It might be more logical to read it in each case statement and parse it locally. **I have already re-architected the DSM code to have a standalone parser being able to parse a byte array**.

This is all information you would need to completely finish this.

kd0aij, 2016-01-07:
@LorenzMeier  first order of business is to get sbus working on master again
We have a regression since beta; sbus input was working then.
I'm looking to see what has changed for fmuV4; let me know if you have any hints

kd0aij, 2016-01-07:
@LorenzMeier The regression for fmuV4 sbus input occurred between v1.1.0 and d314915
I've been unable to compile any of the commits between those two points, though.
It looks like the compilation errors are all real, nothing that a clean will fix, and there are over 2 dozen commits in that range. But with the changes to cmake and QtCreator, I may be confused, so I'll wait for your comments before trying to diagnose the problem.

LorenzMeier, 2016-01-07:
I can't debug this now, but I'm sure a few printfs will quickly let you narrow it down.

kd0aij, 2016-01-07:
Finally got sbus working on beta pixracer in master. Now on to RC scan.

kd0aij, 2016-01-07:
timing looks funny in the sbus decoder. interval between frames is precisely 9msec, 
but the hrt_abstime intevals reported by fmu.cpp vary quite a lot:
[image]
[image]

LorenzMeier, 2016-01-07:
Thats absolutely non-surprising. Why would you assume the system is executing exactly when a frame is received? If you look at the driver you'll see there is no blocking poll call.

Also keep in mind that frames are buffered by the DMA RX buffer and then parsed.

In other words: I would totally expect what you're seeing and there is nothing special about it. The art of a realtime system is to buy exact execution where it matters for relaxing timing constraints where it matters less.

LorenzMeier, 2016-01-07:
On the timing: an overall timing quality estimator for FPV racing would be great. But that's another consideration.

LorenzMeier, 2016-01-07:
Could you send a PR for S.BUS? Thanks for fixing and sorry for the inconvenience caused with it.

kd0aij, 2016-01-07:
will do the PR since it sounds like it's working as expected now
I will need to make sure it wasn't just hardware differences between R7 and R12 though.

kd0aij, 2016-01-08:
Regarding the timing checks, it was my understanding that earlier fixes had something to do with "timing" and I'd also heard that PixRacer reduced control latency.  I'll have to think about why the RC inputs aren't handled using the "poll" mechanism (that misnomer always bothers me).
And minimum latency for control inputs is high on my personal list of requirements, second only to control loop bandwidth, I think.

LorenzMeier, 2016-01-08:
You would need to change the FMU driver back to a thread polling on control topics and FDs for that, and we probably should do that. If / when needed I'll handle that.

closed 2016-01-20 by LorenzMeier